FreeCAD/FreeCAD-addons #172: [Bug] CfdOF link is broken

matsievskiysv, 2020-09-12:
**Describe the bug**
CfdOF link is broken

**To Reproduce**
Steps to reproduce the behavior:
1. Open addon manager
2. Try to find CfdOF
3. Fail

**Expected behavior**
CfdOF to be in the addon list

**Desktop (please complete the following information):**
 - OS: Debian bullseye
 - FreeCAD Version OS:
  Debian GNU/Linux bullseye/sid
  Word size of OS: 64-bit
  Word size of FreeCAD: 64-bit
  Version: 0.18.4.
  Build type: Release
  Python version: 3.8.3
  Qt version: 5.12.5
  Coin version: 4.0.0
  OCC version: 7.4.0
  Locale: English/United Kingdom (en_GB)

matsievskiysv, 2020-09-12:
@luzpaz you updated the link

luzpaz, 2020-09-13:
Yea, trying to figure out the problem. Perhaps because it's a gitlab repo ?

matsievskiysv, 2020-09-13:
Looks like FreeCAD actually parses a github page and looks for the links that end with `.git` (<https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/AddonManager/addonmanager_workers.py#L80-L143>)
Since this is a gitlab repo, there's no link. So, this is an addon manager's bug.

I think that downloading and parsing <https://raw.githubusercontent.com/FreeCAD/FreeCAD-addons/master/.gitmodules> is a way to go.

luzpaz, 2020-09-13:
Indeed. 
4d1ba82fd563
Thanks :wink:

luzpaz, 2020-10-17:
@matsievskiysv is this still broken ?

matsievskiysv, 2020-10-17:
> @matsievskiysv is this still broken ?

Not on dev branch

closed 2020-10-17 by matsievskiysv