FreeCAD/FreeCAD-addons #176: Addons not appearing in list

teratux, 2020-10-15:
**Describe the bug**
The list in the Addon Manager view suddenly appears empty

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Tools/Addon Manager'
2. See error

**Expected behavior**
A filled listing of addons

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: openSuse 15.1
 - FreeCAD VersionOS: 
openSUSE Leap 15.1 (KDE/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22366 (Git) AppImage
Build type: Release
Branch: master
Hash: 34c11fcc63efa2fc72cbc74c75c76e9f19d5c84c
Python version: 3.8.5
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

**Additional context**
Just recently installed this AppImage, all was working great a couple of weeks back, noticed this issue from 2 days ago

luzpaz, 2020-10-15:
Update to the latest 0.19

luzpaz, 2020-10-15:
Related: https://github.com/realthunder/FreeCAD_assembly3/issues/315

teratux, 2020-10-15:
> Update to the latest 0.19 (make sure to stay up to date with it as it changes daily)

To be clear, every time I loose the addon list I'll have to update? Couldn't there be a way to establish dependencies or fix versions of each addon for each FreeCAD version?

luzpaz, 2020-10-15:
For those on 0.18.x there is a workaround:
https://forum.freecadweb.org/viewtopic.php?f=3&t=51160

But better to just migrate to latest 0.19.x

luzpaz, 2020-10-15:
> To be clear, every time I loose the addon list I'll have to update? Couldn't there be a way to establish dependencies or fix versions of each addon for each FreeCAD version?

Addons are downloaded via `git` to your `~/.FreeCAD/` folder. They don't need to be downloaded again unless they have an update. The way the Addon Manager works is that it check via `git` to see if there are new commits to said addon. Not sure what you mean by '_every time I loose the addon list_'

teratux, 2020-10-15:
> Not sure what you mean by '_every time I loose the addon list_'

From the issue I'm having the list suddenly one day appeared empty, so it's something to do clearly with the way FreeCAD fetches the list from the main repository.  The fix you provide is to just update FreeCAD altogether, therefore my question, if this issue happens ('every time I loose the addon list') again will the solution be to once again update FreeCAD?

luzpaz, 2020-10-15:
I see, The answer is yes. But the experimental [Extension Manager](https://github.com/mnesarco/FreeCAD_ExtMan) by @mnesarco may work differently.

mnesarco, 2020-10-16:
I am not sure about the AddonManager, but in [ExtMan](https://github.com/mnesarco/FreeCAD_ExtMan) there are many Extensions lists: 

- One from locally installed Extensions in (~/.FreeCAD) 
- Other from [FreeCAD-addons](https://github.com/FreeCAD/FreeCAD-addons)
- Other from [FreeCAD-macros](https://github.com/FreeCAD/FreeCAD-macros)
- Other from [FreeCAD Wiki](https://wiki.freecadweb.org/)
- User defined repositories

All lists (except local installed) are downloaded from internet and cached in (~/.FreeCAD), can be refreshed by the user.
The local installed list is generated directly from the local dir.

teratux, 2020-10-16:
Ok, I'll check that out when I can, I started an update like you suggested, now I'm stuck here, it doesn't go over 95%, started it twice, or do you recommend a complete release AppImage download?.
[image]

luzpaz, 2020-10-16:
@teratux you made need to download the [appimage manually](https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre)...not sure why it chokes up at 95% for you but it's happened to me too. It may be an upstream github issue.

teratux, 2020-10-16:
> @teratux you made need to download the [appimage manually](https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre)...not sure why it chokes up at 95% for you but it's happened to me too. It may be an upstream github issue.

ok, will do then

jaxxor44, 2020-10-19:
OS: Ubuntu 18.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16146 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.18.4)
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

installed .18 from app center, and the addon list only has the "3D tools" item, nothing else
installed .18 from image, issue still exists
updated to .19 and the issue still exists.
downloaded .19 app image, full list is there WOOT WOOT.

luzpaz, 2020-10-19:
> For those on 0.18.x there is a workaround:
> https://forum.freecadweb.org/viewtopic.php?f=3&t=51160
> 
> But better to just migrate to latest 0.19.x

imakin, 2020-10-20:
> Update to the latest 0.19 (make sure to stay up to date with it as it changes daily)

But the latest stable is still on 0.18, right? I assume more people are still using the 0.18 version as encouraged on freecad's official download page

wulph111, 2020-10-24:
> > For those on 0.18.x there is a workaround:
> > https://forum.freecadweb.org/viewtopic.php?f=3&t=51160
> > But better to just migrate to latest 0.19.x

The linked workaround appears to be a separate issue. Following your proposed python code change, the macros populate but the workbenches remain empty (except the 3d_printing_tools).

Fresh install from the "stable" link on the main page. Moving to development snapshots doesn't seem like a proper recommendation for casual users like myself.

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

grexoit, 2020-10-24:
> **Describe the bug**
> The list in the Addon Manager view suddenly appears empty

Just for statistic, I'm currently facing the same identical problem.
Hope in you.

kikislater, 2020-10-25:
Same for me. Latest app image have this as well but rebuilding with lastest 0.18.4 tag on archlinux is fine ... 
Some new users will face bad experience !

luzpaz, 2020-10-25:
CC @yorikvanhavre

grexoit, 2020-10-25:
@luzpaz I have no experience in forum tools. Should I inform Yorik or did it you for me?
By the way I experienced the same problem in two different environment, FreeCAD 0.18 on windows10 and Freecad 0.18 on Fedora Linux: both have in the addon list only 3D_printing_tool!!!

nice sunday to everyone

luzpaz, 2020-10-25:
A solution was pushed to 0.18.4 branch... but a new release hasn't been cut. Please be patient.
Again, not an orthodox thing to do but we recommend in this instance to upgrade to the latest 0.19 (and stay current with it)  
Most everyone on the forum uses it and we hope it will be released by the end of this year.

Kaiser1401, 2020-11-06:
Add me to this... 
If 0.18 is the last stable and the list disappears without me having changed anything on my side, I also feel that is bad user experience. There clearly must be a change on the remote end, there where the addon manager gets it's list from, that should not be happening. And as the solution is to keep updating to the last development release (every time) it seems this is a reoccuring thing. I see that it's nice to have a central repo for the addons, but if it breaks running applications behaviour there might be need for additions (e.g. just collect frome HEAD if tag <my version +1> has not been set or something)

luzpaz, 2020-11-06:
@Kaiser1401 you are correct, we host the Addon repo on github and recently github has changed some of it's infrastructure. This broke v0.18 addon manager functionality. It was fixed in 0.19 and also in 0.18 but wasn't released. 
Due to different challenges, virtually no effort is going in to keeping 0.18.4 up to date. This is what the devs have signaled. This may be unorthodox and may not be the smoothest user experience, but this is the realm of Free Libre Open Source Software. We pivot hard sometimes. FreeCAD hasn't officially reached v1.0 yet. 

Eventually we're working to move toward this solution: [Extension Manager](https://forum.freecadweb.org/viewtopic.php?f=10&t=50265).  But that takes some time and more effort. Feel free to pitch in.

adrien-n, 2020-11-10:
You can apply https://github.com/FreeCAD/FreeCAD/commit/258f9f1577e71e30f8696b266458df23042eefa5 in order to fix the issue.

I skipped the two subsequent commits ( https://github.com/FreeCAD/FreeCAD/commit/b7ab01e6de6087bf7109b983183598bf0bb4358e and https://github.com/FreeCAD/FreeCAD/commit/734202bc810bf61bd6f66bdc060e6951bd0d134d ) because I wanted to keep changes minimal and avoid copy-paste errors (I use the debian package from testing).

surfound, 2020-12-16:
OS: Ubuntu 16.04 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23323 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19_pre)
Python version: 3.5.2
Qt version: 4
Locale: chinese

I download it yesterday and found the same question. I can’t see the workbench list but I can see the macro list...

zharfanw, 2021-01-19:
[image]

I have solved it.
I Change recplace command in "Program Files/Freecad/Mod/AddonManager.py" at list 571, and replace it with the next list (572)

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Indonesia (en_ID)

zharfanw, 2021-01-19:
> [image]
>
> I have solved it.
> I Change recplace command in "Program Files/Freecad/Mod/AddonManager.py" at list 571, and replace it with the next list (572)
>
> OS: Windows 10
> Word size of OS: 64-bit
> Word [...]

[image]

This image after i modified the python code,

luzpaz, 2021-01-19:
For all those running v0.18.4 please note that there is a way to modify the code so the Addon Manager works. We've backported a fix but github changed their infrastructure again (yes a 2nd time) so we will not be releasing any more backports. The best way to keep abreast of the changes is to use v0.19_pre.

cadsimula-xin, 2021-01-26:
for 0.19, I changed code lines as below. It works for me.

/Mod/AddonManager/addonmanager_utilities.py
[image]

/Mod/AddonManager/addonmanager.py
[image]

shangjiaxuan, 2021-01-31:
Most recent 19_pre (23756) is also having this problem (downloaded from the zip archive). Any suggestion on why it's happening?

shangjiaxuan, 2021-02-01:
> Most recent 19_pre (23756) is also having this problem (downloaded from the zip archive). Any suggestion on why it's happening?

I tried to debug the thing, and it seems to be urlopen method  not working. Forcing a dictionary of NULL in proxy code fixed the problem.

luzpaz, 2021-03-24:
This is a won't fix. Please install v0.19 or provide a patch that users can apply manually.

ozzyman, 2021-05-31:
OK, just had the same issue in the latest stable 0.19 release. Running freecad from a terminal revealed an error in one of the workbench files.

This is the error.:
During initialization the error "invalid syntax (assembly2lib.py, line 59)" occurred in /home/tony/.FreeCAD Mod/FreeCAD_assembly2/InitGui.py

Deleting the offending folder (/home/tony/.FreeCAD Mod/FreeCAD_assembly2) solved the issue. It was probably a leftover from 0.18, but Addon manager really should be able to step over issues like this and just flag the workbench as faulty.

luzpaz, 2021-05-31:
Assembly2 is obsolete

ozzyman, 2021-06-01:
> Assembly2 is obsolete

I'm aware of that. AS I SAID it was probably a leftover from 0.18, which I haven't used for over 18 months. What I wasn't expecting however is that it would cause the upgraded Addon Manager to completely fail to load any addons. I suspect that the reason I could only see 3D printer workbench before I upgraded to the current release of 0.19 is that Addon manager was loading 3D workbench and then failing when it got to Assembly2. "Assembly2" is, of course, alphanumerically subsequent to "3D workbench". 

I posted this so that anyone else who is having the same issue (can only see 3D workbench in addon manager) might be able to solve the problem.

RaphaelWimmer, 2021-06-07:
> for 0.19, I changed code lines as below. It works for me.
[...]
> /Mod/AddonManager/addonmanager.py
> [image]

Thanks. This fix made the FreeCAD 0.19 package in Debian work again. The file to edit on Debian is called `/usr/share/freecad/Mod/AddonManager/AddonManager.py`. (the other suggested edit is neither possible nor necessary in the version currently in Debian (0.19.1).

luzpaz, 2021-06-07:
@ozzyman thanks for the heads up and apologies for being redundant. I'll open a new ticket for the assembly2 addon's presence causing a bug. 

@RaphaelWimmer I checked the 0.19.2 source (latest stable) and couldn't find the code snippet you shared.

RaphaelWimmer, 2021-06-07:
@luzpaz - 
> @RaphaelWimmer I checked the 0.19.2 source (latest stable) and couldn't find the code snippet you shared.

Hmm, the Debian package is at 0.19.1 - however, even the [Debian 0.19.1 sources](https://sources.debian.org/src/freecad/0.19.1+dfsg1-2/src/Mod/AddonManager/AddonManager.py/) don't seem to contain the lines that @ozzyman mentioned. 
However, I'm pretty sure that I installed 0.19.1 (and even verified it).
I'll try to find out more about this issue later. At first glance, it looks like a packaging problem. (I didn't want to reopen this bug but add the solution to my problems to this thread.)

luzpaz, 2021-06-07:
@RaphaelWimmer thanks. Feel free to comment in the issue even though I'm closing it.

closed 2021-06-07 by luzpaz

RaphaelWimmer, 2021-06-08:
Just for closure: for some reason, when I upgraded the freecad Debian package from 0.18 to 0.19, the packaging manager decided to keep the `freecad-runtime` v0.18 package (including the old 'Addon Manager' code) instead of replacing it with the `freecad-common` v0.19 package. I don't understand how this could have happened and cannot reproduce it anymore.

s--d, 2021-07-14:
> Thanks. This fix made the FreeCAD 0.19 package in Debian work again. The file to edit on Debian is called `/usr/share/freecad/Mod/AddonManager/AddonManager.py`. (the other suggested edit is neither [...]

The exact same change in AddonManager.py worked for me on version 0.18.4 (on line 574 instead of 89) to allow the Workbench list to populate, and I was able to install one.  I did not modify addonmanager_utilities.py, however.  YMMV, and of course, we've been advised above to move to, and remain updated on, the latest 0.19 release.