FreeCAD/FreeCAD-addons #177: [New Addon/Workbench]

davidmubernal, 2020-10-16:
Hello!
I would like to add the worbench to the Addon Manager:
https://github.com/davidmubernal/Mechatronic

Thanks!
### Dependencies
- This Addon/workbench runs on:  
  - [x] Most up to date stable release
  - [x] Most up to date development release
  - [x] Python 3 compatible
  - [x] Qt5 compatible
- [ ] Is this backward compatible with any previous version of FC? If so, please mention in the README.md
- [ ] Does this use 3rd party dependencies?  
  - [ ] numpy
  - [ ] scipy
  - Please specify if there are any other dependencies below and in the README.md

### README.md
- [x] Is [Markdown][Markdown-link] friendly
- [ ] Has a **Prerequisite** or **Dependencies** section
- [x] Links to a ``LICENCE`` file (Note: FreeCAD uses LGPL3, most people choose that or GPL3)
- [x] Has a Screenshot and/or a section for screenshots
- [x] Links to a [FreeCAD Forum discussion thread][FreeCAD-Thread  

### Misc.
- [ ] Do you keep a [changelog][Changelog]?  
  If so, where?  
- [ ] Your Addon has an SVG based Logo?
- [ ] Once your Addon/Workbench is accepted, you'll open a PR for FreeCAD master in order to add said Logo in to the FreeCAD Addon Manager dialog? ([Example][Add-Icons-to-Master]), see directions below:   
  1. Copy the SVG to `FreeCAD/src/Mod/AddonManager/Resources/Icons` (Note the formatting of the file name is important).
  2. Edit the `FreeCAD/src/Mod/AddonManager/Resources/AddonManager.qrc` and alphabetically place the path (in the previous step) to the icon of your workbench in the list.  
- [ ] Edit the FreeCAD [External Workbenches][FC-ExternalWB-Wiki] wiki page

### Housekeeping
- [ ] Fixed typos in the code using [`codespell`][Codespell]. We recommend running: `codespell -q 3`
- [x] We recommend linting python code with PEP8. See [coding conventions][Code-conventions] 

### Github Related
- [x] Does your repository use [Release tags][Github-Tags] ?
- [x] Repository has [topics][Github-Topics] which include `FreeCAD` `Workbench` `Addon`  

**Note:** if your addon/workbench code doesn't reside on Github, please state the address where it is mentioned

 
[Markdown-link]: https://guides.github.com/features/mastering-markdown/
[FreeCAD-Thread]: https://forum.freecadweb.org/viewforum.php?f=8
[Changelog]: https://keepachangelog.com/en/1.0.0/
[Add-Icons-to-Master]: https://github.com/FreeCAD/FreeCAD/commit/bd985feef323468380a2e5dd88fb3b7046849826
[FC-ExternalWB-Wiki]: https://www.freecadweb.org/wiki/External_workbenches
[Coding-conventions]: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Fem/coding_conventions.md#python-and-c
[Codespell]: https://github.com/codespell-project/codespell
[Github-Topics]: https://help.github.com/en/articles/classifying-your-repository-with-topics
[Github-Tags]: https://help.github.com/en/articles/creating-releases

luzpaz, 2020-10-16:
@davidmubernal thanks!  
You repo doesn't have [topics](https://help.github.com/en/articles/classifying-your-repository-with-topics) associated with it as requested in the template above

davidmubernal, 2020-10-16:
@luzpaz Thanks for the advise. I add the topics to the repo.

If you see other thinks wrong or that need to be added. Let me know
Thanks

luzpaz, 2020-10-16:
The README is a little sparse. It's important that the README be more informative since it will be displayed in the Addon Manager dialog. Take a look at some of other workbenches through the Addon Manager, like A2+, Assembly4, Sheetmetal etc...

luzpaz, 2020-10-23:
> The README is a little sparse. It's important that the README be more informative since it will be displayed in the Addon Manager dialog. Take a look at some of other workbenches through the Addon Manager, like A2+, Assembly4, Sheetmetal etc...

bump

davidmubernal, 2020-10-24:
I have update the README

luzpaz, 2020-10-25:
Please add a section at the bottom of the README.md for the license like:
  
### License

[GPLv3](LICENSE)

davidmubernal, 2020-10-27:
Thank you, I added the License section right now

commit a4b06e46 (2020-10-30): Adding Mechatronic workbench

Closes #177
closed 2020-10-30 by luzpaz