FreeCAD/FreeCAD-addons #181: [New Addon/Workbench] OSE Piping Workbench

rkrenzler, 2020-11-26:
Could you please add OSE Piping Workbench to the list of addons?
https://github.com/rkrenzler/ose-piping-workbench

### Dependencies
- This Addon/workbench runs on:  
  - [x] Most up to date stable release
  - [x] Most up to date development release
  - [x] Python 3 compatible
  - [x] Qt5 compatible
- [ ] Is this backward compatible with any previous version of FC? If so, please mention in the README.md
- [ ] Does this use 3rd party dependencies?  
  - [ ] numpy
  - [ ] scipy
  - Please specify if there are any other dependencies below and in the README.md
  - [x] Flamingo Workbench, optional

### README.md
- [x] Is [Markdown][Markdown-link] friendly
- [x] Has a **Prerequisite** or **Dependencies** section
- [x] Links to a ``LICENCE`` file (Note: FreeCAD uses LGPL3, most people choose that or GPL3)
- [x] Has a Screenshot and/or a section for screenshots
- [ ] Links to a [FreeCAD Forum discussion thread][FreeCAD-Thread  

### Misc.
- [ ] Do you keep a [changelog][Changelog]?  
  If so, where?  
- [x] Your Addon has an SVG based Logo?
- [x] Once your Addon/Workbench is accepted, you'll open a PR for FreeCAD master in order to add said Logo in to the FreeCAD Addon Manager dialog? ([Example][Add-Icons-to-Master]), see directions below:   
  1. Copy the SVG to `FreeCAD/src/Mod/AddonManager/Resources/Icons` (Note the formatting of the file name is important).
  2. Edit the `FreeCAD/src/Mod/AddonManager/Resources/AddonManager.qrc` and alphabetically place the path (in the previous step) to the icon of your workbench in the list.  
- [x] Edit the FreeCAD [External Workbenches][FC-ExternalWB-Wiki] wiki page

### Housekeeping
- [x] Fixed typos in the code using [`codespell`][Codespell]. We recommend running: `codespell -q 3`
- [x] We recommend linting python code with PEP8. See [coding conventions][Code-conventions] 

### Github Related
- [ ] Does your repository use [Release tags][Github-Tags] ?
- [x] Repository has [topics][Github-Topics] which include `FreeCAD` `Workbench` `Addon`  

**Note:** if your addon/workbench code doesn't reside on Github, please state the address where it is mentioned

 
[Markdown-link]: https://guides.github.com/features/mastering-markdown/
[FreeCAD-Thread]: https://forum.freecadweb.org/viewforum.php?f=8
[Changelog]: https://keepachangelog.com/en/1.0.0/
[Add-Icons-to-Master]: https://github.com/FreeCAD/FreeCAD/commit/bd985feef323468380a2e5dd88fb3b7046849826
[FC-ExternalWB-Wiki]: https://www.freecadweb.org/wiki/External_workbenches
[Coding-conventions]: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Fem/coding_conventions.md#python-and-c
[Codespell]: https://github.com/codespell-project/codespell
[Github-Topics]: https://help.github.com/en/articles/classifying-your-repository-with-topics
[Github-Tags]: https://help.github.com/en/articles/creating-releases

luzpaz, 2020-11-26:
@rkrenzler nice work!!!  
Question, in your README you mention using @oddtopus's Flamingo WB  
> To use all features install Flamingo-Workbench.  

But Flamingo has been superceded by [Dodo WB](https://github.com/oddtopus/dodo) which is the Py3/Qt5 port of Flamingo. That looks like it will be an issue.

yorikvanhavre, 2020-11-26:
Merged with adbaf02
I'm leaving this open for @luzpaz 's question... But I believe Dodo's commands are more or less fully compatible with Flamingo.
Great work on the OSE piping WB!

luzpaz, 2020-11-26:
@rkrenzler please update the README to reflect that the workbench can be installed via the Addon Workbench. Here's an example: https://github.com/Mambix/FreeCAD-yaml-workbench/tree/master#installation

luzpaz, 2020-11-26:
@yorikvanhavre does this mean that dodo can be installed as a dependency and the OSE piping workbench still work? If so, @rkrenzler should mention this in the README as well

yorikvanhavre, 2020-11-26:
This needs to be tested I guess :) But I would suppose yes

luzpaz, 2020-11-26:
> This needs to be tested I guess :) But I would suppose yes

The reason being that Flamingo will raise red flags in the AddonManager due to it being only compatible with Py2. So I think porting to Dodo is priority because it would be very confusing for users which will create more unnecessary questions on the forum 
 
[image]

rkrenzler, 2020-11-26:
@yorikvanhavre, @luzpaz thank you for your feedback!
I will then port the workbench to Dodo with a high priority.

rkrenzler, 2020-11-26:
> @rkrenzler please update the README to reflect that the workbench can be installed via the Addon Workbench. Here's an example: https://github.com/Mambix/FreeCAD-yaml-workbench/tree/master#installation

[v] Thank you. Done.

luzpaz, 2020-12-03:
@rkrenzler is what's the time-frame for porting your code from flamingo to dodo ?

rkrenzler, 2020-12-06:
I hope to do it in January. But it is hard to estimate, because I do not want to break compatibility with Flamingo. I want that all parts created in the old workbench continue to work, if people do not have Dodo installed.

luzpaz, 2020-12-06:
@rkrenzler when is OSE going to update their FreeCAD version? You guys are at 0.16 still, right ?

luzpaz, 2020-12-06:
We need some sort of heads up to the user then in that case. What do you propose ?

rkrenzler, 2020-12-07:
I think that OSE uses mostly 0.18 and 0.19 version of FreeCAD.

I will try to create a workbench which works both with Flamingo and with Dodo but. It will show a warning that the Flamingo is deprecated and suggest to move to Dodo.

rkrenzler, 2020-12-10:
I added Dodo support to the workbench.

This workbench uses Dodo by default. If Dodo is not present it will use Flamingo.

luzpaz, 2020-12-10:
Awsome. @rkrenzler does it notify the user in all cases ?

rkrenzler, 2020-12-12:
It will display a warning, if the user tries to crate a new part.
[image]
If the user will not create anything, she/he will not see the warning.

luzpaz, 2020-12-12:
Excellent. Although at some point we should create some documentation for workbench devs on how to standardize reporting errors in their workbenches. Closing ticket for now.

closed 2020-12-12 by luzpaz