FreeCAD/FreeCAD-addons #183: [Bug] Failed to load Std_AddonMgr

metropt, 2020-12-18:
When I try to run FreeCAD it fails to load the addon manager. I tried to update python packages but I got no success on solve the problem.

`ze@ze:~/Downloads$ ./FreeCAD_0.19-23323-Linux-Conda_glibc2.12-x86_64.AppImage 
FreeCAD 0.19, Libs: 0.19R23323 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCAUBMX1q/usr/Mod/AddonManager/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCAUBMX1q/usr/Mod/Arch/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCAUBMX1q/usr/Mod/Draft/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCAUBMX1q/usr/Mod/Tux/InitGui.py
Please look into the log file for further information
During initialization the error "<method-wrapper '__get__' of getset_descriptor object at 0x7f97a9a75200> returned a result with an error set" occurred in /home/ze/.FreeCAD/Mod/kicadStepUpMod/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in freecad.appimage
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "/tmp/.mount_FreeCAUBMX1q/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/.mount_FreeCAUBMX1q/usr/lib/python3.8/site-packages/freecad/appimage/init_gui.py", line 6, in <module>
    from . update import AppImageUpdaterDialog
  File "/tmp/.mount_FreeCAUBMX1q/usr/lib/python3.8/site-packages/freecad/appimage/update.py", line 4, in <module>
    from PySide2 import QtWidgets, QtGui, QtCore
  File "/home/ze/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "/home/ze/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "/home/ze/.local/lib/python3.8/site-packages/shiboken2/__init__.py", line 30, in <module>
    _init_pyside_extension()
NameError: name '_init_pyside_extension' is not defined
--------------------------------------------------------------------------------
During initialization the error "name '_init_pyside_extension' is not defined" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "/tmp/.mount_FreeCAUBMX1q/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/.mount_FreeCAUBMX1q/usr/lib/python3.8/site-packages/freecad/asm3/init_gui.py", line 3, in <module>
    from .utils import mainlogger as logger
  File "/tmp/.mount_FreeCAUBMX1q/usr/lib/python3.8/site-packages/freecad/asm3/utils.py", line 10, in <module>
    import FreeCAD, FreeCADGui, Part, Draft
  File "/tmp/.mount_FreeCAUBMX1q/usr/Mod/Draft/Draft.py", line 38, in <module>
    import Draft_rc
  File "/tmp/.mount_FreeCAUBMX1q/usr/Mod/Draft/Draft_rc.py", line 5, in <module>
    from PySide2 import QtCore
  File "/home/ze/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "/home/ze/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "/home/ze/.local/lib/python3.8/site-packages/shiboken2/__init__.py", line 30, in <module>
    _init_pyside_extension()
NameError: name '_init_pyside_extension' is not defined
--------------------------------------------------------------------------------
Unknown command 'Std_AddonMgr'
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/tmp/.mount_FreeCAUBMX1q/usr/Mod/Start/StartPage/StartPage.py", line 29, in <module>
    from . import TranslationTexts
  File "/tmp/.mount_FreeCAUBMX1q/usr/Mod/Start/StartPage/TranslationTexts.py", line 24, in <module>
    from PySide import QtGui
  File "/tmp/.mount_FreeCAUBMX1q/usr/Ext/PySide/__init__.py", line 2, in <module>
    from PySide2 import __version__
  File "/home/ze/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "/home/ze/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "/home/ze/.local/lib/python3.8/site-packages/shiboken2/__init__.py", line 30, in <module>
    _init_pyside_extension()
<class 'NameError'>: name '_init_pyside_extension' is not defined
name '_init_pyside_extension' is not defined
`

`OS: Ubuntu 20.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23323 (Git) AppImage
Build type: Release
Branch: master
Hash: 512d5c6141aec52b6eecc67370336a28fde862a6
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
`

luzpaz, 2020-12-18:
I can't reproduce on  
```
Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23323 (Git) AppImage
Build type: Release
Branch: master
Hash: 512d5c6141aec52b6eecc67370336a28fde862a6
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

Also this is not the right address for this ticket. Please open a forum thread to discuss with the community.

closed 2020-12-18 by luzpaz

metropt, 2020-12-18:
@luzpaz can you share your PySide2 package version?

luzpaz, 2020-12-18:
My system says `5.15.2-1` but not sure what version is installed with the AppImage.

luzpaz, 2020-12-19:
@metropt Ran an `--appimage-extract` and checked  `usr/lib/pkgconfig/pyside2.pc`  
Current AppImage bundles pyside2 v5.13.2

metropt, 2021-02-05:
```
ze@ze-XPS-13-9380:~/Downloads/squashfs-root/usr/lib/pkgconfig$ cat pyside2.pc 
prefix=/home/travis/build/FreeCAD/FreeCAD-AppImage/conda/linux/AppDir/usr
exec_prefix=/home/travis/build/FreeCAD/FreeCAD-AppImage/conda/linux/AppDir/usr
libdir=lib
includedir=/home/travis/build/FreeCAD/FreeCAD-AppImage/conda/linux/AppDir/usr/include/PySide2
typesystemdir=/home/travis/build/FreeCAD/FreeCAD-AppImage/conda/linux/AppDir/usr/share/PySide2/typesystems
gluedir=/home/travis/build/FreeCAD/FreeCAD-AppImage/conda/linux/AppDir/usr/share/PySide2/glue
pythonpath=/home/travis/build/FreeCAD/FreeCAD-AppImage/conda/linux/AppDir/usr/lib/python3.8/site-packages

Name: PySide2
Description: Support library for Python bindings of Qt5-based libraries.
Version: 5.13.2
Libs: -L${libdir} -lpyside2.cpython-38-x86_64-linux-gnu
Cflags: -I${includedir}
Requires: shiboken2

```

metropt, 2021-02-05:
The error persist on last AppImage

metropt, 2021-02-06:
After analyse the error log I found that I have a python binary and packages installed on ~/.local `"/home/ze/.local/lib/python3.8/site-packages/PySide2/__init__.py",` that is used by the AppImage, maybe because of some environment variable? After remove pyside2 and shiboken2 from the .local it is working.

qingfengxia, 2021-04-15:
this issue is caused by user has installed diff version pyside2 via pip, which has higher priority

https://forum.freecadweb.org/viewtopic.php?f=4&t=57786&p=496662#p496662

beevabeeva, 2021-06-26:
Had the same issue. Thanks for helping.

Solution 
```pip3 uninstall pyside2
pip3 uninstall shiboken2
```