FreeCAD/FreeCAD-addons #189: [FEATURE] Addon Manager runs a post uninstall script to cleanup any remaining files

luzpaz, 2021-02-21:
**Is your feature request related to a problem? Please describe.**  
Sometimes certain addons install files that don't get addressed/removed on uninstall of the addon. A good example of this is the Extension Manager. @mnescaro, the ExtMan dev writes:  
>  The problem with `/tmp` is that it gets removed on system restarts and the cache contains a lot of files, so i prefer to keep it on a more stable location. If there is a way to instruct the AddnonManager to do additional cleanups on Addons uninstalls i will use that. But there is no such functionality as far as i know. 

**Describe the solution you'd like**  
Addon Manager should have post-uninstall phase where it can run an auxiliary script that could clean up non-standard install files.

**Describe alternatives you've considered**  
Haven't considered any

**Additional context**
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=50265&start=60#p480208

mnesarco, 2021-02-21:
One easy solution is that the AddonManager do seach for a file inside the Addon directory and run it if found before uninstall.

ie:

ExtMan addon can provide this file:

`$HOME/.FreeCAD/Mod/ExtMan/freecad/extman/uninstall.py`

So the search pattern could be something like:

`$ADDONS_DIR/$ADDON_NAME/freecad/*/uninstall.py
`

chennes, 2022-01-24:
As suggested by @mnesarco : https://github.com/FreeCAD/FreeCAD/pull/5432

chennes, 2022-01-28:
Merged, closing. See documentation at https://wiki.freecadweb.org/Workbench_creation#Distribution

closed 2022-01-28 by chennes