FreeCAD/FreeCAD-addons #190: [Bug] Chinese local users reporting that 0.19.1 Addon Manager is not displaying any addons

luzpaz, 2021-05-01:
**NOTE: PLease open tickets regarding individual addons/workbenches in their respective developer's source code repository. This repository is used to track FreeCAD Addon Manager based bugs** 

**Describe the bug**  
Chinese locale users have malfunctioning Addon Manager that doesn't display any addons  
Thread (in chinese): https://forum.freecadweb.org/viewtopic.php?f=40&t=55532

**To Reproduce**  
(To be translated)

**Expected behavior**  
Addon manager displays addons

**Screenshots**  
[image]

**Desktop (please complete the following information):**

[image]

**Additional context**  
Possible unicode issue I imagine.

SilentCA, 2021-07-14:
It may be the network problem. 
Check whether you can access "https://raw.githubusercontent.com/FreeCAD/FreeCAD-addons/master/addonflags.json", where FreeCAD gets addon list.
I fixed the network problem by editing hosts file, then FreeCAD works fine.

hughchin95, 2021-11-14:
you are right,I can not link this  website .  I am trying to use VPN to solve  it.  I will come  back here if I succeed.

hughchin95, 2021-11-14:
I solved it by add hosts of github.com to C:\Windows\System32\drivers\etc\hosts

chennes, 2022-01-01:
This then looks like a local DNS issue, right? Can we close?

closed 2022-01-03 by luzpaz