FreeCAD/FreeCAD-addons #192: Presence of obsolete Assembly2 addon on local machine will break Addon Manager

luzpaz, 2021-06-07:
OK, just had the same issue in the latest stable 0.19 release. Running freecad from a terminal revealed an error in one of the workbench files.

This is the error.:
During initialization the error "invalid syntax (assembly2lib.py, line 59)" occurred in /home/tony/.FreeCAD Mod/FreeCAD_assembly2/InitGui.py

Deleting the offending folder (/home/tony/.FreeCAD Mod/FreeCAD_assembly2) solved the issue. It was probably a leftover from 0.18, but Addon manager really should be able to step over issues like this and just flag the workbench as faulty.

_Originally posted by @ozzyman in https://github.com/FreeCAD/FreeCAD-addons/issues/176#issuecomment-851201668_

chennes, 2022-01-01:
It should be straightforward for Addon Manager to check if an obsolete or Py2 addon is installed (in fact, with the new Addon Manager it is already doing so, it just doesn't take action). We should discuss on the forum what an appropriate course of action is in this case.

chennes, 2022-02-13:
Mostly as a note to myself so I remember: this isn't really anything to do with Addon Manager.  FreeCAD should wrap workbench init in a try block and provide some useful feedback, as well as do something like offer to uninstall the offending workbench.