FreeCAD/FreeCAD-addons #193: [FEATURE] Run requirements.txt

Mambix, 2021-06-20:
**Is your feature request related to a problem? Please describe.**
I'm the creator of yaml-workspace which relies on yaml python package.
At the moment you can not use the workspace as freecad complains yaml module is not installed. I have yaml module installed on my linux system but freecad apparently uses it's own python distribution.

**Describe the solution you'd like**
If the github project contains requirements.txt file the add on manager should make sure those modules will be available for the workspace to work properly.

**Describe alternatives you've considered**
A description of how we can manually add modules to internal python instalation.

**Additional context**
OS: Pop!_OS 20.10 (KDE/plasma); 
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Unknown
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.0
Locale: English/United States (en_US)

luzpaz, 2021-06-23:
@looooo whats the workaround for this?

chennes, 2022-01-01:
I'm pinging myself on this: the only real solution is for FreeCAD to have the option of installing the required packages when they aren't found. That way if whether we are using the system Python, or our own packaged version, things will work properly.

chennes, 2022-01-12:
I'm sort of doing this (in spirit anyway) -- with https://github.com/FreeCAD/FreeCAD/pull/5376 I am reading in requirements.txt (along with metadata.txt and package.xml) and consolidating their python packages information, and then installing them via pip if they aren't already on the system. It's not actually giving pip the requirements.txt file, to prevent some types of malware distribution it just pulls out the dependencies and installs them individually, and ignores version information and any extra commands in the file. The requirements.txt file for the mentioned addon parses without issue and logs a requirement on PyYaml. I'll add that to our allowed installs list.

chennes, 2022-01-22:
Closing this now that the PR is question has been merged. If anyone requires more sophisticated requirements.txt support please contact me.

closed 2022-01-22 by chennes