FreeCAD/FreeCAD-addons #200: Tracking migration to updated metadata schema

luzpaz, 2022-01-15:
## Background
Thanks to @chennes we have a really robust way to categorize and systematize addons/macros/preference packs. So this thread will track the progress of getting addon devs to migrate to it. 

* Info about the schema is at: https://wiki.freecad.org/Package_Metadata
* Example of an implemeted schema: TBD

### Workbenches
GDML https://github.com/KeithSloan/GDML/issues/76

DeepSOIC, 2022-01-19:
> \<date>
>
> REQUIRED
>
> The date of the last update, in the format YYYY-MM-DD or YYYY.MM.DD. 

Am i now supposed to update it upon every commit to my workbench? Because that sounds like a bit of a git merge conflict generator, apart from being an easy-to-forget red-herring...

i'm sorry if it's a stupid question or not the right place to discuss it. I didn't instantly find it at the forum at least.

chennes, 2022-01-19:
The idea isn't to update it every commit, but every time you change the version number.

DeepSOIC, 2022-01-19:
@chennes oh, wait, do i have to change the version number for every commit too?

chennes, 2022-01-19:
You can do whatever you want with the version number, I don't think we want to get in the business of telling developers how they have to do that. I'm working on some new code now to better handle displaying the last commit date, and that will also enable better update handling for both git and non-git users. So don't worry too much about it right now.

davesrocketshop, 2022-01-19:
Rocket workbench updated

luzpaz, 2022-01-20:
@chennes should we fork @triplus's repos to the FreeCAD organization and maintain them ?

chennes, 2022-01-20:
It would probably be better if there were a specific human being who was interested in maintaining some or all of them. I haven't seen triplus around in a year or two, but I don't know the story there.

chennes, 2022-01-20:
On a different note, Pyrate has the metadata file in their official repo on Debian Salsa. I have not ever tried to have two different submodules with the same name, though, so I don't know how to handle that. 0.19.3 supports Salsa now, but nothing before that does. I don't know when we want to pull the trigger on making that shift.

luzpaz, 2022-01-21:
I think I may be able to find some volunteers. I'll start a tracking ticket.

luzpaz, 2022-01-21:
> I have not ever tried to have two different submodules with the same name, though, so I don't know how to handle that.  

Oy vey, here comes the complexity.

luzpaz, 2023-04-13:
I guess we should close this ticket?

closed 2023-08-05 by luzpaz