FreeCAD/FreeCAD-addons #201: [Tracking] Various themes/stylesheets for FreeCAD master and Link branch

luzpaz, 2022-01-19:
* [Original themes](https://github.com/pgilfernandez/FreeCAD_stylesheets) by @pgilfernandez (bundled in to master)
* [Blender theme](https://github.com/pgilfernandez/FreeCAD_Blender-stylesheet) by @pgilfernandez
* [ProDark](https://github.com/turn211/ProDark-FreeCAD-theme) by @turn211 (Bundled in to master)
* [Behave dark theme](https://github.com/Chrismettal/FreeCAD-Behave-dark) by @Chrismettal (master)
* [freecad-stylesheets](https://github.com/MathiasStadler/freecad-Stylesheets) by @MathiasStadler 
* [Plane stylesheet](https://github.com/wfpaisa/Plane-freecad) by @wfpaisa
* [Dracula dark theme](https://github.com/dracula/freecad) by @dracula
* [Prof-CAD Dark/Light Themes](https://github.com/OBKF/Prof-CAD by @OBKF
* 
--------

Link/LinkStage3/LinkDaily by @realthunder
* [OficineRobotica's Preference Presets](https://github.com/oficinerobotica/FreeCAD-Preferences-Presets) by @oficinerobotica's (Link/LS3/Daily)  
* [VertUI](https://github.com/StudioPetrikas/FreeCADLink_VertUI) by @StudioPetrikas (LinkDaily)
* [CadWin](https://github.com/rmTizi/CadWin} by @rmTizi (Link)

luzpaz, 2022-01-31:
Thanks @Chrismettal! for creating a preference pack from your theme https://github.com/FreeCAD/FreeCAD-addons/pull/207

CC 
@turn211
@MathiasStadler
@wfpaisa
@dracula
@OBKF

Also, even though y'all made you themes for LS3/LD, nonetheless take note as when @realthunder gets his Preference editor changes merged upstream it would be still appropriate to distribute your themes as preference packs :wink: 
@oficinerobotica
@StudioPetrikas
@rmTizi

luzpaz, 2022-04-27:
Tracking ticket for elements in the FreeCAD UI that aren't themeable: https://github.com/FreeCAD/FreeCAD/issues/6805

turn211, 2022-05-19:
Made ProDark Pack https://github.com/turn211/ProDarkThemePreferencePack but can't get it to work with manual install. 
https://forum.freecad.org/viewtopic.php?f=34&t=68771&sid=7fdbe3ef5a9fe6f05e8def16651f5328

chennes, 2022-05-19:
@turn211 I submitted a PR to fix a typo in your package.xml file.

luzpaz, 2022-07-14:
Dracula Preference pack has been merged in #228  
Thanks @chrismettal !

luzpaz, 2022-07-15:
Merged [Extreme ProDark](https://github.com/MisterMakerNL/ExtremeProDark) Preference pack by @MisterMakerNL (#229)

luzpaz, 2022-08-22:
Added @macdroid53's https://github.com/macdroid53/BasicMech dark theme to the list

luzpaz, 2022-10-02:
Heads up stylesheet devs, the default blue color in property editor can be modified now. Note: a restart of FreeCAD is needed to see the changes. Details: https://forum.freecadweb.org/viewtopic.php?p=630033#p630033

pgilfernandez, 2022-10-03:
> Heads up stylesheet devs, the default blue color in property editor can be modified now. Note: a restart of FreeCAD is needed to see the changes. Details: https://forum.freecadweb.org/viewtopic.php?p=630033#p630033
> 
> Edit: thanks to @wwmayer and @MisterMakerNL

So nice to hear it... I have been fighting with link colors so many hours without luck, but now I see that the solution was related with restarting FreeCAD to see the changes... grrr

luzpaz, 2022-10-03:
> I have been fighting with link colors so many hours without luck, but now I see that the solution was related with restarting FreeCAD to see the changes... grrr

That is a frustrating feeling, for sure. Maybe worthy of opening a feature request to try to make it work in real-time. Or we should let folks know, in CAPS, that thus is necessary for certain stylesheet elements

MisterMakerNL, 2022-10-03:
It's weird that if I switch from my test theme with orange link colors to my published theme I still get blue links without restarting. After restart they are orange again.

luzpaz, 2022-10-07:
Added [ExtremeProLight](https://github.com/MisterMakerNL/ExtremeProLight) by @MisterMakerNL to OT

luzpaz, 2022-10-15:
Has anyone here seen this https://github.com/dprojects/scanObjects by @dprojects ? Could this be useful aid helping with stylesheet themeing ?

[image]

[image]

[image]

MisterMakerNL, 2022-10-16:
I checked it and I think I could find some usefull information. I found this about the constrains list that we have issues in theming. Now only decipher it too Qt. 
```
        <Property name="Constraints" type="Sketcher::PropertyConstraintList" status="1">
            <ConstraintList count="25">
                <Constrain Name="" Type="1" Value="0" First="0" FirstPos="2" Second="1" SecondPos="1" Third="-2000" ThirdPos="0" LabelDistance="10" LabelPosition="0" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                <Constrain Name="" Type="1" Value="0" First="8" FirstPos="2" Second="2" SecondPos="1" Third="-2000" 
            </ConstraintList>
```
Also checked QtGui but that was more general information about the Qt options there are. Or I missed something.

luzpaz, 2023-01-19:
Heads up to theme/Pref-pack devs: https://github.com/FreeCAD/FreeCAD/pull/7888 is a PR by realthunder that adds an overlay to the Gui doc widgets. 

https://github.com/FreeCAD/FreeCAD/pull/7888/files#diff-a2f9cae4d1847cafe1a9a5765834b9b90c9b761bbf92c4df9f53c1dd0fc001dc shows the stylesheet additions that will be available to theme once this is merged.

maxwxyz, 2023-01-30:
cannot wait for [OficineRobotica's Preference Presets](https://github.com/oficinerobotica/FreeCAD-Preferences-Presets) by @oficinerobotica's in main FreeCAD

luzpaz, 2023-09-10:
https://github.com/glassondesignstudio/GdsLightBlue/issues/14

luzpaz, 2025-12-02:
Nordic theme: https://github.com/erroronline1/NordicFC  
* https://github.com/erroronline1/NordicFC/issues/1