FreeCAD/FreeCAD-addons #202: [Tracking] Finding maintainers for abandoned Addons

luzpaz, 2022-01-21:
Starting this ticket to track finding maintainers for the following repos:

* https://github.com/triplus/CubeMenu/
* https://github.com/triplus/PieMenu
* https://github.com/triplus/TabBar
* https://github.com/triplus/ShortCuts
* https://github.com/triplus/IconThemes
* https://github.com/triplus/Launcher

mdkus, 2022-09-15:
Hi luzpaz,
It's no problem for me to take over triplus' other modules as well. It would make sense for me to maintain all triplus' modules. What do you think?
Regards
Michael

luzpaz, 2022-09-21:
PieMenu maintainership has been transferred to @mdkus in [6a8f294](https://github.com/luzpaz/FreeCAD-addons/commit/6a8f294f5254bfd2d79fd06a4224e4fb7c2f9a6e)

hasecilu, 2024-01-16:
Someone uses CADQuary module? at least for me installing it give me problems with Plot, [see my comment](https://github.com/jmwright/cadquery-freecad-module/pull/154#issuecomment-1894439763).

Palmstroemen, 2024-01-30:
I'm currently working on a [WorkbenchOrganizer](https://github.com/Palmstroemen/WB_Organizer) which is based on triplus TabBar. 
As my WorkbenchOrganizer extends the TabBars functionality significantly I gave it a new name. I also ported it to PySide2. 
So my WorkbenchOrganizer is a kind of 'TabBar on Steroids' and could be seen as a replacement. If I only would succeed to get it into the AddOn-Manager.

Palmstroemen, 2024-01-30:
I'm also working on the ShortCuts AddOn from triplus at the moment.
[image]

So maybe in a few weeks I can provide a ShortCuts-AddOn on steroids as well.

hasecilu, 2024-01-31:
Hi @luzpaz, recently **Grub** user have taken over the maintainership of the PieMenu WB, already a lot of improvements have been made, previous maintainer **mdkus** have agreed via forum: https://forum.freecad.org/viewtopic.php?t=84101&start=80

To reflect the change I made this PR: https://github.com/FreeCAD/FreeCAD-addons/pull/285

luzpaz, 2024-02-01:
@chennes :point_up:

shaise, 2024-02-03:
I think triplus's selector toolbar is much better then the tab bar: https://github.com/triplus/SelectorToolbar
and should be considered instead

dosas, 2024-05-30:
@luzpaz what about https://github.com/execuc/LCInterlocking ? The PRs seem to be unreviewed?

luzpaz, 2024-11-13:
SearchBar maintainer has changed 
* https://github.com/FreeCAD/FreeCAD-addons/issues/330

luzpaz, 2025-01-11:
Dodo has been deprecated (https://github.com/FreeCAD/FreeCAD-addons/pull/347) in favor of Quetzel