FreeCAD/FreeCAD-addons #209: [FEATURE] Ability to copy just the keyboard shortcuts out of a specific Preference Pack in to ones current theme

luzpaz, 2022-02-25:
**Is your feature request related to a problem? Please describe.**
Preference Packs provide the ability to customize the FreeCAD UI in a multitude of ways. But there exists a scenario where a user might already like the esthetic UI/UX setup they want but want to import just keyboard shortcuts from another Preference Pack. This feature would provide a way to solely import a Preference Pack's assigned keyboard shortcuts (nothing else)

**Describe the solution you'd like**
In the Preferences of the Preference Pack menu clicking on the a specific Preference pack opens up a confirmation dialog asking if the user is sure they want to install chosen pref pack. In this dialog create a checkbox that when enabled  solely imports said pref pack's keyboard shortcuts.  

**Describe alternatives you've considered**
A macro that pulls shortcuts from prefpacks and implements them in the user's user.cfg. Or manually opening up a pref pack and copy/pasting (xml) shortcuts in to one's own `user.cfg`

**Additional context**
Forum mention: https://forum.freecadweb.org/viewtopic.php?style=1&p=574337#p574337

chennes, 2023-01-08:
The problem I see with this is that keyboard shortcuts are just one type of preference one might want to separate out: I can think of lots of subcategories that might get split out, and that quickly becomes untenable in the user interface. Absent some significant demand for the automated "parting out" of preference packs, my inclination is to rely on addon auhors to partition their Addons as they see fit.

luzpaz, 2023-04-23:
@chennes this feature was recently added to core, right ?

chennes, 2023-04-23:
No, the feature that was added was the ability to save your keyboard shortcuts into a new pack, not to extract just a subset of an existing pack (which is what is needed to implement this ticket's suggestion).