FreeCAD/FreeCAD-addons #213: [FEATURE] 'Add custom repository' button in the Addon Manager that links to Preferences -> Addon Manager -> Addon Manager preferences

luzpaz, 2022-04-17:
**Is your feature request related to a problem? Please describe.**
Just a UX convenience  

**Describe the solution you'd like**
[image]
which opens the Preferences -> Addon Manager -> Addon Manager preferences
 
[image]

**Describe alternatives you've considered**
Some static text to orient the user. But a button seems smarter.

**Additional context**
Not sure if there are technical issues with this for example, does addon manager dialog need to be closed in order to open the 'Addon Manager preferences' and to then see the changes made to the preferences?

chennes, 2022-09-21:
I'm about at the limit of the number of buttons I can comfortably cram down in the lower left corner, and your technical concern is right -- the Addon Manager reads its preferences only when it is launched, not continuously while it is running, so the Addon Manager would have to restart when the preferences changed. I'm not sure I see any great advantage to adding this button considering the difficulty on the technical side.

luzpaz, 2022-09-21:
Yea, closing.

closed 2022-09-21 by luzpaz