FreeCAD/FreeCAD-addons #215: [FEATURE] Add python version metadata to `package.xml`

luzpaz, 2022-04-28:
**Is your feature request related to a problem? Please describe.**
Addon Manager will make it easier for everyone if it checked minimum/maximum python version for an addon. Example: https://forum.freecadweb.org/viewtopic.php?p=591793#p591793

**Describe the solution you'd like**
Per @chennes 
> I think the way to implement it is to allow you to specify a dependency on Python, and then write the code to support the version info (which the metadata supports, but I haven't actually implemented yet). So you'd have something like:
```
<depend version_gte="3.7">python</depend>
```
> This won't work right now because the Addon Manager will assume that "python" is some sort of pip-installable package. I'd just have to manually code in support for it, but I've already done that for Plot, so it's not a huge departure.

chennes, 2022-09-23:
Implemented in https://github.com/FreeCAD/FreeCAD/commit/94498816bdf4147cd48a7057e6b048c544b3226c

closed 2022-09-23 by luzpaz

chennes, 2022-09-24:
(A side note that I am reconsidering this implementation, it causes some backward compatibility issues, so I haven't documented this yet).