FreeCAD/FreeCAD-addons #216: [Bug]

sslupsky, 2022-04-28:
**Describe the bug**
Addon manager does not refresh when running latest 0.20

**To Reproduce**
Steps to reproduce the behavior:
1. Tools->Addon manager
2. Select any addon in the list (example below I selected kicadstepup and 3D printing tools )
3. Window is empty

[image]

[image]

**Expected behavior**
Window should display git info

**Desktop (please complete the following information):**
```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.28765 (Git)
Build type: Release
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * kicadStepUpMod 10.10.4
```

luzpaz, 2022-04-28:
Hey @ipatch, do you mind trying to reproduce on macOS (HEAD) ?

ipatch, 2022-04-28:
yeah it'll take me a few.

@sslupsky are you launching freecad from cli or are using a `FreeCAD.app` bundle?

if building from source, please produce commands to replicate build, please and thank you. 🙏

sslupsky, 2022-04-28:
@ipatch I launched the app bundle.  I downloaded the latest nightly (weekly?) build of 0.20 earlier today (28765)

sslupsky, 2022-04-28:
My previous install of 0.20 (build 27078) works fine.

ipatch, 2022-04-28:
@sslupsky 

it seems you're using the conda release

https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-28765-OSX-x86_64-conda.dmg

ill have to check and see if i exp the same issue if i build a bundle using cmake with homebrew provided packages

chennes, 2024-02-28:
I have received no further reports of this behavior -- if it reappears it is a packaging issue that we'll have to address in the specific executable that is being used.

closed 2024-02-28 by chennes