FreeCAD/FreeCAD-addons #218: [New Addon/Workbench]

SHaiZen25, 2022-04-30:
### Dependencies
- This Addon/workbench runs on:  
  - [x] Most up to date stable release
  - [x] Most up to date development release
  - [x] Python 3 compatible
  - [x] Qt5 compatible
- [x] Is this backward compatible with any previous version of FC? If so, please mention in the README.md
- [x] Does this use 3rd party dependencies?  
  - nine
  - docutils 0.18
  - blinker 1.4
  - nine 1.1.0

### README.md
- [x] Is [Markdown][Markdown-link] friendly
- [x] Has a **Prerequisite** or **Dependencies** section
- [x] Links to a ``LICENCE`` file (Note: FreeCAD uses LGPL3, most people choose that or GPL3)  
- [x] Has a Screenshot and/or a section for screenshots
- [x] Links to a [FreeCAD Forum discussion thread][FreeCAD-Thread

### Misc.
- [ ] Do you keep a [changelog][Changelog]?  
  If so, where?  
- [x] Your Addon has an SVG based Logo?
- [ ] Once your Addon/Workbench is accepted, you'll open a PR for FreeCAD master in order to add said Logo in to the FreeCAD Addon Manager dialog? ([Example][Add-Icons-to-Master]), see directions below:   
  1. Copy the SVG to `FreeCAD/src/Mod/AddonManager/Resources/Icons` (Note the formatting of the file name is important).
  2. Edit the `FreeCAD/src/Mod/AddonManager/Resources/AddonManager.qrc` and alphabetically place the path (in the previous step) to the icon of your workbench in the list.  
- [ ] Edit the FreeCAD [External Workbenches][FC-ExternalWB-Wiki] wiki page

### Housekeeping
- [x] Fixed typos in the code using [`codespell`][Codespell]. We recommend running: `codespell -q 3`
- [x] We recommend linting python code with PEP8. See [coding conventions][Code-conventions] 

### Github Related
- [x] Does your repository use [Release tags][Github-Tags] ?
- [ ] Repository has [topics][Github-Topics] which include `FreeCAD` `Workbench` `Addon`  

**Note:** if your addon/workbench code doesn't reside on Github, please state the address where it is mentioned

 
[Markdown-link]: https://github.com/QuentinTournier40/AnimationFreeCAD/blob/main/README.md
[FreeCAD-Thread]:  https://forum.freecadweb.org/viewtopic.php?f=8&t=36299&start=280
[Changelog]: x
[Add-Icons-to-Master]: https://github.com/QuentinTournier40/AnimationFreeCAD/blob/main/icons/clapCinema.svg
[FC-ExternalWB-Wiki]: x
[Coding-conventions]: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Fem/coding_conventions.md#python-and-c
[Codespell]: https://github.com/codespell-project/codespell
[Github-Topics]: animation, visual-scripting
[Github-Tags]: animation, visual-scripting

luzpaz, 2022-07-28:
Which workbench is this?

luzpaz, 2023-01-19:
@AndreasCottet ping

chennes, 2024-03-14:
Closing as inactive.

closed 2024-03-14 by chennes