FreeCAD/FreeCAD-addons #221: [FEATURE] Test Github connectivity button within Addon Manager UI

luzpaz, 2022-05-06:
**Is your feature request related to a problem? Please describe.**    

Users are confused if Addon Manager contains a bug or if there are connectivity problems. This solution can help them rule out the latter. 

**Describe the solution you'd like**

A simple button in the Addon Manager UI that when invoked will test the connection with Github.  
Something that automates the below script and returns success or failure to the user. 

```python
import urllib.request
r = urllib.request.urlopen("https://api.github.com/zen")
r.read()
```

**Describe alternatives you've considered**

n/a

**Additional context**

Forum request: https://forum.freecadweb.org/viewtopic.php?style=3&p=593545#p593545

chennes, 2022-05-06:
I'm not sure the intent here: the Addon Manager already does this check automatically, and pops up an error message when it is launched without connectivity. The main GUI does not show at all if it cannot connect.

luzpaz, 2022-05-06:
> I'm not sure the intent here: the Addon Manager already does this check automatically, and pops up an error message when it is launched without connectivity.  

Users aren't grokking this. I've been seeing messages on different social media about confusion around this. Also for someone to test connectivity would they need to close and re-open Addon Manager each time? A button allows for checking connection immediately 

BTW, the connectivity function should also test other repos like gitlab, salsa, and codeberg (etc...)

chennes, 2022-05-06:
It's checking Github because that's where FreeCAD-Addons is located -- if we can't access that, we can't show the list of Addons. I'm extremely hesitant to add another button to the user interface that is only useful if you lose (or regain) internet connectivity while you have the Addon Manager window open. Every added button makes the software harder to use, so we need to be judicious with our choices here: it has to add enough value to counteract the clutter it contributes to the interface.

luzpaz, 2022-05-07:
OK. I understand your concern. I'll close this ticket for now.

closed 2025-05-12 by PhoneDroid