FreeCAD/FreeCAD-addons #222: addon manager: display addon version and last commit date

FreeCAD-Bug-Importer, 2022-05-25:
Issue imported from https://tracker.freecad.org/view.php?id=4801

* **Reporter:** roipoussiere
* **Date submitted:** 12/11/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

This is useful information, that can be used to quickly know if an add-on is still in development.

It can be located for instance, just under the warning / info box above the add-on description.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 1/8/2022

# Discussion from Mantis ticket

### Comment by openBrain 2021-12-14 11:05
roipoussiere : please proceed according our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by first discussing your request in our forum. Thx

---

### Comment by roipoussiere 2021-12-14 17:02
suggestion message on the forum: https://forum.freecadweb.org/viewtopic.php?p=553702#p553702

---

### Comment by chennes 2021-12-16 19:41
Addon Manager is in the process of being redesigned. I will eventually incorporate this suggestion into the display of whether an addon needs to be updated.

---

### Comment by chennes 2022-01-08 04:09
Half of this is done (the display of version info), but displaying the last commit info is more complex. Discussion topic here: https://forum.freecadweb.org/viewtopic.php?f=8&t=65109

luzpaz, 2022-05-25:
Transferred ticket to https://github.com/FreeCAD/FreeCAD-addons  
@chennes when you get a chance, what's left here?

chennes, 2024-02-28:
This is now in the Addon Manager.

closed 2024-02-28 by chennes