FreeCAD/FreeCAD-addons #225: fatal: No url found for submodule path 'FCGear' in .gitmodules

luzpaz, 2022-06-13:
Trying to do a repo update, i encounter this error when running this one liner (it avoids checking out submodules who's default branch is not 'master')

```shell
git submodule foreach 'case $name in MnesarcoUtils|SlopedPlanesMacro|Design456|OpticsWorkbench|Beltrami|Trails|SearchBar|AnimationFreeCAD) ;; *) git checkout master ;; esac'
```

Does this error have to do with afd5107f8eafbd4c046e72c855372026abb76e9e ?

chennes, 2022-06-13:
I don't know - the Addon Manager doesn't use the submodules feature of git, it just parses the .gitmodules file. I don't know what git does if the file changes and it already has those submodules checked out.

luzpaz, 2022-08-13:
This is still a problem. Have not been able to update the submodules since the .gitmodules file was altered in https://github.com/FreeCAD/FreeCAD-addons/commit/afd5107f8eafbd4c046e72c855372026abb76e9e

PR #234 (2022-09-11): bugfix: fix broken FCGears reference (fixes #225)
Fixes #225
commit 50133cc9 (2022-09-11): bugfix: fix broken FCGears reference (fixes #225)
closed 2022-09-11 by chennes