FreeCAD/FreeCAD-addons #226: [New Stylesheet] ExtremeProDark

MisterMakerNL, 2022-07-04:
It's just a style-sheet no need for a lot of text for explaining [link](https://github.com/MisterMakerNL/ExtremeProDark), its a in my opinion better version of the prodark.
Also the pro dark dev doesn't seem to react anymore or update his code.
https://github.com/MisterMakerNL/ExtremeProDark

luzpaz, 2022-07-04:
Doesn't have a package.xml file: https://wiki.freecadweb.org/Package_Metadata

MisterMakerNL, 2022-07-05:
@luzpaz I added it!

luzpaz, 2022-07-05:
Nice, giving props to @turn211 and @studiopetrikas :wink:

luzpaz, 2022-07-05:
@MisterMakerNL this stylesheet works on master, correct?

Chrismettal, 2022-07-11:
Don't we need to package it as a complete PreferencePack to provide it via the addon manager? 
The original [ProDark](https://github.com/turn211/ProDarkThemePreferencePack) follows the packaging format as well and I don't see a reason to stray from the norm here. In the past stylesheets were included in FreeCADs source directly, and now we supply them via the addon manager as PreferencePacks.

See also the [guide to PreferencePack structure](https://wiki.freecadweb.org/Preference_Packs#Preference_Pack_structure)

MisterMakerNL, 2022-07-11:
I just copied it from pro-dark. :)

Chrismettal, 2022-07-11:
Can you make sure your pack conforms to the structure specified completely, and then try to manually install your pack with the addon manager in the same way as described here, by manually adding your packs repo as an addon source:

https://github.com/dracula/freecad/issues/2#issuecomment-1179737161

If that works, you can open up a PR to this repo, adding your pack, and testers can try your pack out the same way before merging.

luzpaz, 2022-07-15:
Closing in favor of #229

closed 2022-07-15 by luzpaz