FreeCAD/FreeCAD-addons #230: Alternate OpenSCAD Importer

KeithSloan, 2022-08-18:
Addon Manager does not seem to allow install via config anymore, so would like to add here
https://github.com/KeithSloan/OpenSCAD_Alt_Import

Does the icon have to be SVG?

### Dependencies
- This Addon/workbench runs on:  
  - [x ] Most up to date stable release
  - [x ] Most up to date development release
  - [ x] Python 3 compatible
  - [ x] Qt5 compatible
- [ x] Is this backward compatible with any previous version of FC? If so, please mention in the README.md
- [ No] Does this use 3rd party dependencies?  
  - [ ] numpy
  - [ ] scipy
  - Please specify if there are any other dependencies below and in the README.md

### README.md
- [ ] Is [Markdown][Markdown-link] friendly
- [ ] Has a **Prerequisite** or **Dependencies** section
- [ ] Links to a ``LICENCE`` file (Note: FreeCAD uses LGPL3, most people choose that or GPL3)
- [ ] Has a Screenshot and/or a section for screenshots
- [ ] Links to a [FreeCAD Forum discussion thread][FreeCAD-Thread  

### Misc.
- [ ] Do you keep a [changelog][Changelog]?  
  If so, where?  
- [ ] Your Addon has an SVG based Logo?
- [ ] Once your Addon/Workbench is accepted, you'll open a PR for FreeCAD master in order to add said Logo in to the FreeCAD Addon Manager dialog? ([Example][Add-Icons-to-Master]), see directions below:   
  1. Copy the SVG to `FreeCAD/src/Mod/AddonManager/Resources/Icons` (Note the formatting of the file name is important).
  2. Edit the `FreeCAD/src/Mod/AddonManager/Resources/AddonManager.qrc` and alphabetically place the path (in the previous step) to the icon of your workbench in the list.  
- [ ] Edit the FreeCAD [External Workbenches][FC-ExternalWB-Wiki] wiki page

### Housekeeping
- [ ] Fixed typos in the code using [`codespell`][Codespell]. We recommend running: `codespell -q 3`
- [ ] We recommend linting python code with PEP8. See [coding conventions][Code-conventions] 

### Github Related
- [ ] Does your repository use [Release tags][Github-Tags] ?
- [ ] Repository has [topics][Github-Topics] which include `FreeCAD` `Workbench` `Addon`  

**Note:** if your addon/workbench code doesn't reside on Github, please state the address where it is mentioned

 
[Markdown-link]: https://guides.github.com/features/mastering-markdown/
[FreeCAD-Thread]: https://forum.freecadweb.org/viewforum.php?f=8
[Changelog]: https://keepachangelog.com/en/1.0.0/
[Add-Icons-to-Master]: https://github.com/FreeCAD/FreeCAD/commit/bd985feef323468380a2e5dd88fb3b7046849826
[FC-ExternalWB-Wiki]: https://www.freecadweb.org/wiki/External_workbenches
[Coding-conventions]: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Fem/coding_conventions.md#python-and-c
[Codespell]: https://github.com/codespell-project/codespell
[Github-Topics]: https://help.github.com/en/articles/classifying-your-repository-with-topics
[Github-Tags]: https://help.github.com/en/articles/creating-releases

luzpaz, 2022-08-18:
@chennes do you mind taking a look at this?

FYI, fixed the markdown in the checkboxes list.

chennes, 2022-08-19:
It lists a license file named "LICENSE" in its metadata, but I don't see that file.

KeithSloan, 2022-08-20:
Should now be added okay.

chennes, 2023-01-08:
Are all steps complete on this now?

KeithSloan, 2023-01-08:
Think so, Luzpaz closed Aug last year see[https://github.com/KeithSloan/OpenSCAD_Alt_Import/issues?q=is%3Aissue+is%3Aclosed](url)

closed 2023-01-08 by chennes