FreeCAD/FreeCAD-addons #243: [Bug]Remove automatic py-slsv install for Assembly3 workbench

CBtiger, 2022-10-30:
**Describe the bug**
AddOn Manager does install the python solverspace binding for the ASM 3 workbench into the subfolder "AdditionalPythonPackages". The problem with that is that different Python versions are installed in the same place causing conflicts when different FreeCAD versions are used (e.g. different AppImages). 
Since the Asm3 does install its required py-slvs but it own when its not found this AddOnManager feature is not required any more and should be removed.

Here the relevant section from a Realthunder answer from one of the references
 `.. the new AddonManager installs the dependency in .local/share/FreeCAD/AdditionalPythonPackages which may cause problem if you are alternating AppImages that pack different version of Python, because py_slvs is a native package. Please go in this directory and manually remove all py_slvs directories. Then start FreeCAD, and if you've already have asm3 installed, switch to the workbench and it will prompt to install py_slvs. Try and see if this can get it installed. If you haven't installed asm3 (or removed it previously), when prompt to install dependency, choose Ignore and try.`

References:
 https://github.com/realthunder/FreeCAD_assembly3/issues/1282
 https://github.com/FreeCAD/FreeCAD-Bundle/issues/139

**To Reproduce**
Steps to reproduce the behavior:
1. Remove all oy AdditionalPythonPackages
2. uninstall Asm3 workbench
3. install it again. AddOn Manager will ask to install py-slvs
4. It creates the files in AdditionalPythonPackages

**Expected behavior**
A clear and concise description of what you expected to happen.
1. Remove all oy AdditionalPythonPackages
2. uninstall Asm3 workbench
3. install it again. No AddOnManager intervention
4. Select Assembly3 WB
5. a popup will ask to install py-slvs
6. the report view will show a successful install. Its in a different folder than the AddonManager install

Note
7. A "failed" window may popup. This is a Asm3 bug and can be ignored if the Report view shows a successful install

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - FreeCAD Version: tested in Linux, but should apply to windows as well.
[code]
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.29457 (Git) AppImage
Build type: Release
Branch: development/toponaming
Hash: 18ff9bcf2610a04e07b87fa077abedeb4b2b2761
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * TabBar
  * dodo
  * Assembly3 0.11.3
  * DynamicData 2.46.0
  * Assembly4 0.12.3
  * Defeaturing
  * kicadStepUpMod 10.14.5
  * ExplodedAssembly
  * Manipulator 1.4.9
  * MeshRemodel 1.8919.0
  * ArchTextures
  * Trails 2022.1.0
  * ThreadProfile 1.84.0
  * Help 1.0.3
  * SelectorToolbar
  * Behave-Dark-Colors 0.1.1
  * LCInterlocking
  * InventorLoader 1.3.0
  * InventorUbuntuInstallScript
  * 3D_Printing_Tools
  * PieMenu 1.1.10
  * Animation
  * workfeature
  * lattice2 1.0.0
  * MakerWorkbench
  * Reporting
  * Curves 0.5.2
  * sheetmetal 0.2.58
  * IconThemes
  * fasteners 0.4.1
  * Render 2022.2.0
  * PreferencesPresets
  * Design456 0.0.1
  * exportOffline.sh
  * Pyramids-and-Polyhedrons
  * CurvedShapes 1.0.4
  * offline-documentation 1.0.0-alpha
  * Glass
  * Sample
  * fcgear 1.0.0
  * FCGear
  * BIM 2021.12.0
[/code]

**Additional context**
n/a

chennes, 2022-10-30:
This isn't really an Addon Manager bug -- @realthunder should remove "py-slvs" as a dependency if he wants to handle installing it himself. In the meantime, when the Addon Manager asks you if you want to install the dependency, click "Ignore" instead of "Install".

closed 2022-10-30 by chennes

realthunder, 2022-11-01:
So just to be clear, @chennes , is it true that the `package.xml` dependency is meant only for non-native Python package?

chennes, 2022-11-01:
Right now it is meant just to tell the Addon Manager to install the package if it can't find it on the system already.

chennes, 2022-11-01:
I see your point, I wasn't thinking about binary packages -- I'll change the Addon Manager to version that directory.