FreeCAD/FreeCAD-addons #247: Remove Old Workbench.

iplayfast, 2023-01-14:
The RemBench addon is apparently messing up the menu system in FreeCad (File and View menus can't be clicked on). 
After figuring out that it was the RemBench that was responsible, I realized that it is actually obsolete. It's purpose was to trace the different workbenches as you moved across different tabs. 
FreeCad does this without the workbench installed, so I propose removing it from the add on list, and possibly adding a popup warning for people who have it installed.

luzpaz, 2023-01-14:
Please link to original ticket this was found in. TIA!

luzpaz, 2023-01-15:
Original issue: https://github.com/FreeCAD/FreeCAD/issues/7965

chennes, 2023-01-15:
I have marked it obsolete as of 0.20 -- should that be 0.21 instead? Note that right now the AM doesn't actively prompt for uninstallation, this feature just prevents it from being installed new.

luzpaz, 2023-01-16:
@iplayfast :point_up:

iplayfast, 2023-01-17:
This problem occurred around  Thu Jun 16 09:03:24 2022 +0200. I'm not sure of the version number at that time. 0.20 is probably safest.

Syres916, 2023-02-01:
Surely the root cause is any Addon that adds a menu, both PieMenu and Rembench do this (Accessories) and both give the same symptoms, so is there an alternative method for creating the menu in Python that doesn't cause the symptoms or find out why the core code has started this regression??