FreeCAD/FreeCAD-addons #251: [New Addon/Workbench] Taack PLM

Taack, 2023-02-10:
Addon repository: https://github.com/Taack/taack-plm-freecad

### Dependencies
- This Addon/workbench runs on:  
  - [x] Most up to date stable release
  - [ ] Most up to date development release
  - [x] Python 3 compatible
  - [x] Qt5 compatible
- [ ] Is this backward compatible with any previous version of FC? If so, please mention in the README.md
- [ ] Does this use 3rd party dependencies?  
  - [ ] numpy
  - [ ] scipy
  - Please specify if there are any other dependencies below and in the README.md

### README.md
- [x] Is [Markdown][Markdown-link] friendly
- [ ] Has a **Prerequisite** or **Dependencies** section
- [ ] Links to a ``LICENCE`` file (Note: FreeCAD uses LGPL3, most people choose that or GPL3)
- [ ] Has a Screenshot and/or a section for screenshots
- [x] Links to a [FreeCAD Forum discussion thread][FreeCAD-Thread  

### Misc.
- [x] Do you keep a [changelog][Changelog]?  
  If so, where?  In the forum thread
- [x] Your Addon has an SVG based Logo?
- [ ] Once your Addon/Workbench is accepted, you'll open a PR for FreeCAD master in order to add said Logo in to the FreeCAD Addon Manager dialog? ([Example][Add-Icons-to-Master]), see directions below:   
  1. Copy the SVG to `FreeCAD/src/Mod/AddonManager/Resources/Icons` (Note the formatting of the file name is important).
  2. Edit the `FreeCAD/src/Mod/AddonManager/Resources/AddonManager.qrc` and alphabetically place the path (in the previous step) to the icon of your workbench in the list.  
- [ ] Edit the FreeCAD [External Workbenches][FC-ExternalWB-Wiki] wiki page

### Housekeeping
- [ ] Fixed typos in the code using [`codespell`][Codespell]. We recommend running: `codespell -q 3`
- [ ] We recommend linting python code with PEP8. See [coding conventions][Code-conventions] 

### Github Related
- [ ] Does your repository use [Release tags][Github-Tags] ?
- [ ] Repository has [topics][Github-Topics] which include `FreeCAD` `Workbench` `Addon`  

**Note:** if your addon/workbench code doesn't reside on Github, please state the address where it is mentioned

 
[Markdown-link]: https://guides.github.com/features/mastering-markdown/
[FreeCAD-Thread]: https://forum.freecadweb.org/viewforum.php?f=8
[Changelog]: https://keepachangelog.com/en/1.0.0/
[Add-Icons-to-Master]: https://github.com/FreeCAD/FreeCAD/commit/bd985feef323468380a2e5dd88fb3b7046849826
[FC-ExternalWB-Wiki]: https://www.freecadweb.org/wiki/External_workbenches
[Coding-conventions]: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Fem/coding_conventions.md#python-and-c
[Codespell]: https://github.com/codespell-project/codespell
[Github-Topics]: https://help.github.com/en/articles/classifying-your-repository-with-topics
[Github-Tags]: https://help.github.com/en/articles/creating-releases

chennes, 2023-02-10:
See https://github.com/Taack/taack-plm-freecad/pull/1

Taack, 2023-02-11:
@chennes if you want to test this workbench, I updated the instructions in the website. Steps to get it up and running are described there https://forum.freecad.org/viewtopic.php?p=659957#p659957.

Regards

Taack, 2023-02-12:
I cannot fork the FreeCAD master ATM, I get a "This repository is temporarily unavailable." error... 

I think I made something wrong when trying to create the pull request...

chennes, 2023-02-13:
Sorry for the bad instructions, they are out-of-date. You do not need to have your icon added to the main repository, that instruction is obsolete. I have updated it in the template.

Taack, 2023-02-15:
@chennes I updated the Workbench. It should not move starting from release https://github.com/Taack/taack-plm-freecad/releases/tag/v20.2.2.

I also added to the release files the server binary. You can launch it on Mac/RPi or your workstation using java **17**. But FreeCAD appimage must be present in your home folder and be set executable.

```
$ ls ~/FreeCAD_0.20.2-2022-12-27-conda-Linux-x86_64-py310.AppImage
FreeCAD_0.20.2-2022-12-27-conda-Linux-x86_64-py310.AppImage
```

In order to launch the server, just type:

```
java -jar server-0.2.jar
```

The database is in the `./prodDb.mv.db` file, the uploaded files will be stored under `~/intranetFiles` by default. I never tested this binary under Windows, but it should work modulo model rendering. On Mac it should work as expected (modulo the `dot` executable path used to represent graphs).

You can modify default settings:

* Path the Intranet store its files: `java -Dintranet.root=/home/test/someDir -jar server-0.2.jar`
* Server port: `java -Dserver.port=9444 -jar server-0.2.jar`
* Database url: `java -DdataSource.url='jdbc:h2:./prodDb2;LOCK_TIMEOUT=10000;DB_CLOSE_ON_EXIT=FALSE' -jar server-0.2.jar`
* Freecad appimage path:`java -Dplm.freecadPath=/somePath...  -jar server-0.2.jar`
* Dot exe path: `java -Dexe.dot.path=/somePath...  -jar server-0.2.jar`

You can combine those options. Will do some doc one day about them (see https://github.com/Taack/intranet/blob/main/server/grails-app/conf/application.yml for an exhaustive list).

It is OK for you to include the link in **FreeCAD-addons** ?

Regards

Taack, 2023-02-15:
Just do a pull request..
https://github.com/FreeCAD/FreeCAD-addons/pull/253

closed 2023-09-10 by luzpaz