FreeCAD/FreeCAD-addons #257: Deprecate ModernUI

luzpaz, 2023-04-16:
Per @MisterMakerNL in https://forum.freecad.org/viewtopic.php?p=670868#p670868

MisterMakerNL, 2023-04-16:
I personally also had issues with ModernUI. It didn't uninstall correctly and screwed up all my addons.
There is a macro to reset it but as I remember it didn't worked correctly and I had to delete my user config.
That is also the reason this addon should probably be removed it screws up the complete UI, it's not like most other addons where bugs are pretty enclosed issues that only occure when the addon is run.

To clarify because it feels like I'm bashing modernUi. The addon is freaking cool and I can't be mad on Hakkan for abandoning it, he even did a fundraiser to support the development, he really did his best too continue working on it. But there was just not enough support. So sadly it needs to put in the Freezer, untill someone wants to update it.

luzpaz, 2023-04-16:
> To clarify because it feels like I'm bashing modernUi.   

That didn't come across for me. Appreciate the heads up.

chennes, 2023-04-16:
Does it work in 0.20, but not in 0.21, or does it fail in 0.20 as well?

998Pete, 2023-04-28:
I have to say I like it and the only issue I have had, that I am aware of, is outlined in the forum
I had no problem deleting it or reloading it when I was trying to establish the cause of my issue so it seems different people (different OS's?) have varying degrees of success with removing it.

My issue was as described here....
"· Load up FreeCAD
· Go to “Part Design”
· Start a new document
· Click on “Create Body”
· Click on “Create Sketch”
· Select “Plane”
· And now FreeCAD is in “Sketcher”…

Except when I do this mine stays in “Part Design”…it doesn’t go to “Sketcher” unless I select “Sketcher” from the Menu."

This is the only issue that I am aware of however as I am basically a complete novice it may be there are other issues that may be more complex or serious.

It is a shame as I like the look and that I can have a description next to each Icon, which for me is useful, at least until I instinctively recognise the function of each, which may be a long way away!

If there is another UI AddOn that works and is similar, with descriptions next to the icons, it would be good to hear about it...

PhoneDroid, 2026-04-02:
Unindexed.

closed 2026-04-02 by PhoneDroid