FreeCAD/FreeCAD-addons #262: [Feature Request] Expose UI Color settings to the user

NXTler, 2023-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.11.3, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * Curves 0.6.12
  * CfdOF 1.23.1
  * TabBar
  * IconThemes
  * Glass
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Core

### Problem description

Hello all,
The colors of any UI can drastically change the feel and usability of it, and every person prefers different colors for their personal usage; hence, I think it's very important that the colors are easily changeable.
In FreeCAD version 0.20, the process of changing the UI's colors is quite tedious. Custom.css files have to be created, which are too long and complicated for the average user who just wants to change the tree's background, for example.
More color options should be exposed through the settings, like they have already been for the start menu.

What are your thoughts about this?

### Anything else?

Here is a screenshot taken from the settings of Solid Edge as an example:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-07-13:
@MisterMakerNL lets compile a list of settings that an addon like this would be able to manipulate?

MisterMakerNL, 2023-08-21:
Axeia has made a nice macro where you can see the preference colors: https://github.com/Axeia/FreeCAD-Color-Fetcher/tree/main

luzpaz, 2023-08-21:
Do you have a quick screenshot to share ?

NXTler, 2023-08-22:
A ReadMe has been added to the project with a screenshot.

luzpaz, 2023-09-10:
What is left to do here ?

NXTler, 2023-09-10:
The Color-Alterator is already great, to be honest. The only feature I miss is being able to set the color of backgrounds (from the model tree for example). Should I move this to the macro?

luzpaz, 2023-09-11:
> Should i move this to the macro?  

Can you clarify what you mean here ?

NXTler, 2023-09-11:
Does this feature request still applies to FreeCAD, or should it be moved to the Color-Agragator?

luzpaz, 2023-09-24:
Probably moved to the Color-Aggregator

NXTler, 2023-09-24:
[Moved to Color-Aggregator](https://github.com/Axeia/FreeCAD-Color-Alterator/issues/3), I close this issue for now.

closed 2023-09-24 by NXTler