FreeCAD/FreeCAD-addons #267: [FEATURE] List Addons by Popularity

brittanyb, 2023-07-29:
Addons provide some extremely powerful features to FreeCAD, but users have no easy way to gauge usefulness of those features without exploring for themselves.

By providing a list order in terms of popularity, it becomes a great deal more clear to those without advanced experience.

Some potential options for gauging popularity:
- Installation/uninstallation count - Consider only verifying upon a completed download.
- Thumbs up/thumbs down system.
- User review/star rating.
- Github stars.

Github stars have the most safety from manipulation considering it expedites the tracking mechanism away from FreeCAD to a much bigger web host. Unfortunately, it's also the least active method, but might be worth considering.

For the others, consider the following:
- No security checks - Might be acceptable if popularity manipulation isn't a concern.
- Basic IP address tracking - Prevents the most basic of attacks, some privacy concerns associated with this.
- Cookies - Same as basic IP address tracking, but already used by FreeCAD.
- Hardware-based identification - Please don't.
- Some kind of integration with the FreeCAD forums? - There are already verified user accounts there, but this is opening up a can of worms. Would only be useful if integration of forums/addon manager has a lot of other features that could be useful.

chennes, 2023-07-29:
This is a work in progress right now, along with a sort by last updated date. We're going to discuss various "popularity" strategies at the [FreeCAD Hackathon](https://blog.freecad.org/2023/07/27/freecad-hackathon-discord/) in a couple of weeks.

brittanyb, 2023-07-29:
Oh cool, thanks for letting me know, and that's nice to hear. :)

chennes, 2024-02-28:
The Addon Manager now provides sorting by GitHub stars (which is a sort of "popularity") as well as an arbitrary "Score" that can be provided by anyone who likes.

closed 2024-02-28 by chennes