FreeCAD/FreeCAD-addons #269: [Bug] Many gitmodules are without .git extension, leading to them not showing up properly

knipknap, 2023-08-07:
**Describe the bug**

Many addons show without descriptions:

[image]

What they all have in common is that they are listed in [.gitmodules](https://github.com/FreeCAD/FreeCAD-addons/blob/master/.gitmodules) without the ".git" at the end.

Any modules that have .git appended show up properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the addon manager
2. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33721 (Git) Snap 730
Build type: Release
Branch: master
Hash: 89d529ab7b0e447e635b44f4f1a8e329c016fedd
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * better-tool-library.backup1691415579.7971573 0.9.4 (Disabled)
  * better-tool-library 0.9.4
  * better-tool-library.backup1691352110.3884366 0.9.4 (Disabled)
  * Manipulator 1.5.0
  * DynamicData 2.46.0
  * toSketch 1.0.1
  * fasteners 0.4.59
  * lattice2 1.0.0
  * Assembly3 0.12.0
  * sheetmetal 0.2.63
  * CycloidGearBox
  * better-tool-library.backup1690119799.0963817 0.9.0 (Disabled)
  * Assembly4 0.50.2
[/code]

chennes, 2023-08-08:
The Addon Manager strips off the .git if it is present, so if that is indeed the problem, it's a bug in the code.

knipknap, 2023-08-11:
Since you just accepted the merge for my library (#272) I have another data point: Looks like just fixing it in .gitmodules doesn't automatically make it appear in this repository either.

Perhaps someone needs to "git submodule update" and push the result to this repository?

chennes, 2023-08-11:
Yes, that is what updates the list on GitHub. We don't do it all that often because it doesn't really affect anything in FreeCAD.

knipknap, 2023-08-11:
I see. And I have to correct myself: My BTL module IS now showing correctly, I just had to refresh the cache. So I think it kind of proofs that the missing .git extension is the cause, one way or another.

chennes, 2024-02-28:
I've modified the submodules file to never include the .git extension, to remove any confusion.

closed 2024-02-28 by chennes