FreeCAD/FreeCAD-addons #274: [Bug] Addon Manager fails to install addon

knipknap, 2023-08-11:
**Describe the bug**

The addon manager fails to download or install an addon, with a Git error message.

**To Reproduce**

1. Open Addon Manager
2. Search for "better-tool-library"
3. Click it, try to install.
4. A popup appears with the following error message:

```
Failed to install btl
Git returned a non-zero exit status: 1
Called with: /snap/freecad/733/usr/bin/git checkout main

Returned stderr:
error: pathspec 'main' did not match any file(s) known to git
```

**Expected behavior**

Installation should work.

**Screenshots**

[image]

**Desktop (please complete the following information):**

[code]
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33789 (Git) Snap 734
Build type: Release
Branch: master
Hash: 22b173d0e27cbb641661d6844512e02020a9890a
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * Manipulator 1.5.0
  * DynamicData 2.46.0
  * btl
  * toSketch 1.0.1
  * fasteners 0.4.59
  * lattice2 1.0.0
  * Assembly3 0.12.0
  * sheetmetal 0.2.63
  * CycloidGearBox
  * Assembly4 0.50.2
[/code]

**Additional context**

This happened after the snap update today.
I double checked the url in .gitmodules in this repository, but can't see any errors.

knipknap, 2023-08-11:
Interestingly, after this error, AddonManager now claims that it is installed (even after a restart), but I do not see it working.

chennes, 2023-08-12:
The error is probably a bad branch name someplace. I’ll look for it.

luzpaz, 2023-08-12:
Is this the fix (https://github.com/FreeCAD/FreeCAD-addons/commit/faf9bc55048ea45c801861d9b3c47f1433655cb9) that solves the problem ?

knipknap, 2023-08-12:
Unfortunately no. Even after this, and clearing the cache, I get this problem.

chennes, 2024-02-28:
This is working for me in the current Weekly of FreeCAD. @knipknap do you have a moment to confirm?

knipknap, 2024-02-28:
Yes, it works right now. Thank you!

closed 2024-02-28 by chennes