FreeCAD/FreeCAD-addons #275: [Bug] `pip install --user` no longer supported in debian trixie

khimaros, 2023-08-22:
**Describe the bug**
i believe https://github.com/realthunder/FreeCAD_assembly3/issues/1317 may actually be a more general issue with AddonManager, since i believe it is responsible for installing deps.

**To Reproduce**
Steps to reproduce the behavior:
1. install debian trixie (testing)
2. install the Assembly3 addon
3. open the workspace and select "Yes" to install py-slvs

**Expected behavior**
installation should succeed

**Desktop (please complete the following information):**
 - OS: debian trixie
 - FreeCAD Version: 0.20.2

chennes, 2023-08-22:
Thanks for the report -- presumably the best solution is for FreeCAD to run its Python interpreter in a virtual environment (eliminating the need for the --user option).

khimaros, 2023-08-28:
@chennes i believe that is the recommended approach now.

chennes, 2024-02-28:
I'm going to close this issue, it's now a FreeCAD repo issue: https://github.com/FreeCAD/FreeCAD/issues/12202

closed 2024-02-28 by chennes