FreeCAD/FreeCAD-addons #278: [New Addon/Workbench] OpenDark preference pack

obelisk79, 2023-09-27:
I would like my Preference Pack "OpenDark" added to the manager. I have verified the repo works with addon-manager as a custom source.

[image]

### Dependencies
- This Addon/workbench runs on:  
  - [x] Most up to date stable release
  - [x] Most up to date development release
  - [x] Python 3 compatible
  - [x] Qt5 compatible
  - [x] Qt5 and Qt6 compatible (uses "import PySide" rather than "import PySide2")
- [x] Is this backward compatible with any previous version of FC? If so, please mention in the README.md
- [ ] Does this use 3rd party dependencies?  
  - [ ] numpy
  - [ ] scipy
  - Please specify if there are any other dependencies below and in the README.md and package.xml files

### README.md
- [x] Is [Markdown][Markdown-link] friendly
- [ ] Has a **Prerequisite** or **Dependencies** section
- [x] Links to a ``LICENCE`` file (Note: FreeCAD uses LGPL2.1, most people choose that or GPL3)
- [x] Has a Screenshot and/or a section for screenshots
- [ ] Links to a [FreeCAD Forum discussion thread][FreeCAD-Thread]

### Misc.
- [x] Do you keep a [changelog][Changelog]?  
  If so, where? GitHub Commits  
- [x] Your Addon has an SVG based Logo?
- [x] Your Addon has a [package.xml file][Package_Metadata]?
- [ ] Edit the FreeCAD [External Workbenches][FC-ExternalWB-Wiki] wiki page

### Housekeeping
- N/A Fixed typos in the code using [`codespell`][Codespell]. We recommend running: `codespell -q 3`
- N/A We recommend linting python code with PEP8. See [coding conventions][Code-conventions] 

### Github Related
- [ ] Does your repository use [Release tags][Github-Tags] ?
- [ ] Repository has [topics][Github-Topics] which include `FreeCAD` `Workbench` `Addon`  

**Note:** if your addon/workbench code doesn't reside on Github, please state the address where it is mentioned

PaddleStroke, 2023-09-27:
I think you perhaps want to set the type to theme so that it appears inside the theme combobox ?

obelisk79, 2023-09-27:
> You perhaps want to set the type to theme so that it appears inside the theme combobox ?

I'm not sure how to do that, I didn't see it documented on the wiki.

PaddleStroke, 2023-09-27:
Just add  `<type>Theme</type>` to the preferencepack in the xml

```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<package format="1" xmlns="https://wiki.freecad.org/Package_Metadata">
 <name>OpenDark</name>
 <version>2023.09.27b</version>
 <date>2023.09.27</date>
 <description>An accessible and coordinated Dark theme for FreeCAD</description>
 <maintainer email="nomail@freecad.org">Obelisk79</maintainer>
 <license file="LICENSE">GPLv3</license>
 <url type="repository" branch="main">https://github.com/obelisk79/OpenDark</url>
 <icon>resources/icons/OpenDark.svg</icon>

<content>
  <preferencepack>
    <name>OpenDark</name>
    <type>Theme</type>
    <description>An accessible and coordinated Dark theme for FreeCAD based on the wonderful Open-Color ui palette</description>
    <tag>2023</tag>
    <tag>theme</tag>
    <tag>dark</tag>
    <file>OpenDark.qss</file>
  </preferencepack>
  <workbench>
    <name>QtNormalizer</name>
    <description>Sets Qt Widget visual style to normalize appearance across disparate compute r platforms</description>
    <classname>QtNormalizer</classname>
  </workbench>
</content>
</package>
```

obelisk79, 2023-09-27:
I added that line, but it didn't remove the pref pack from the list and add it to the theme dropdown as it should.

luzpaz, 2023-09-27:
Issue with license: https://github.com/obelisk79/OpenDark/issues/4

obelisk79, 2023-09-27:
Thank you for pointing that out. I've made the necessary correction. This has been an educational experience :)

luzpaz, 2023-09-27:
@chennes I've given feedback and changes have been implemented. i'll leave it to you to pull the trigger to add to the Addon repo.

chennes, 2023-09-27:
https://github.com/FreeCAD/FreeCAD-addons/pull/279

closed 2023-09-27 by chennes

luzpaz, 2023-09-28:
Congrats @obelisk79! Thank you for your contribution. Thanks chennes for the assist.