FreeCAD/FreeCAD-addons #308: Update the submodules so they are up to date with the HEAD of each addon

luzpaz, 2024-09-11:
See https://gist.github.com/luzpaz/c450aba42444d2d69ba75309f4cd1b60

I used to do this a regular basis but started running into issue. I don't recall what exactly. Maybe we can automate this process ?

luzpaz, 2024-10-25:
I think we should do this before 1.0 release. When people view the addon repo and link to any specific addon, it links to the revision it was at the last time we updated submodules. They don't know that they need to explicitly click on refresh or HEAD to get the most up to date version of the repo. To limit confusion, we can update submodules to avoid this.

chennes, 2024-10-25:
The long-term solution is to stop using submodules like this, since it's a sort of abuse of that system and doesn't work as one would expect. So I wouldn't spend much time on writing automation for it.

luzpaz, 2024-10-25:
Ok, that makes sense. In that case, what is a longterm solution ?

chennes, 2024-10-25:
Probably a JSON file instead of .gitmodules (at any rate, I just ran an update on them), via:

```
git submodule foreach 'git pull origin master || :'
git submodule foreach 'git pull origin main || :'
```
(obviously operating on the assumption that those are the only two branches we need concern ourselves with).

closed 2024-10-25 by chennes

luzpaz, 2024-11-13:
@chennes After changing maintainers for SearchBar (#https://github.com/FreeCAD/FreeCAD-addons/issues/330) I cant seem to run `git submodule` commands from github codespace. 
[image]

Do you mind re-running ?