FreeCAD/FreeCAD-addons #327: Pictures from readme not showing in the addon manager of FreeCAD 1.0

APEbbers, 2024-10-25:
I noticed that with the new FreeCAD 1.0, this is not shown properly. Probaly due to HTML syntaxes. It shows properly with FreeCAD 0.21, 
Have you noticed something like this? The issue persists even when all HTML is removed from the readme.

The pictures below are from the Ribbon UI addon. I had this issue earlier with the TitleBlock WB. Then some code was written to remove HTML. But I think the addon should show the readme as shown in github itself.

FC 1.0:
[image]

FC 0.21:
[image]

luzpaz, 2024-10-25:
Confirmed on:  
```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

chennes, 2024-10-26:
Use relative paths in your image URLs (and yes, Qt's Markdown parser does not understand very much HTML, so your best bet is to use pure markdown in your README).

luzpaz, 2024-10-27:
> Qt's Markdown parser does not understand very much HTML

It understands the collapsed div though right?

chennes, 2024-10-29:
Basically, any HTML support is "at your own risk" -- Qt claims it supports "GitHub-flavored" Markdown, but in my experience you're far better off just keeping it as simple as possible.

APEbbers, 2024-10-29:
I've removed the HTML and the information about other install options. Instead i've added a link to the wiki page "installation".
I've also changed the paths to relative paths:
[image]

is this the correct way? The pictures are still not showing

hasecilu, 2024-10-29:
The only differences I see with other working READMEs is that you don't have text inside the brackets `[]`. 

I use: `![Example of some FreeGrid components](img/screen.png "Example of some FreeGrid components")`, The text between parentheses makes a tooltip to appear on GitHub.

APEbbers, 2024-10-29:
Yes, that solved it! thanks for the help!

closed 2024-10-29 by APEbbers