FreeCAD/FreeCAD-addons #349: [New Addon/Workbench]

Pegoku, 2025-01-17:
repo : https://github.com/Pegoku/freecad-wakatime

### Dependencies
- This Addon/workbench runs on:  
  - [x] Most up to date stable release
  - [x] Most up to date development release
  - [x] Python 3 compatible
  - [] Qt5 compatible  (uses FreeCADGui, so not sure)
  - [] Qt5 and Qt6 compatible (uses "import PySide" rather than "import PySide2")
- [ ] Is this backward compatible with any previous version of FC? If so, please mention in the README.md
- [x] Does this use 3rd party dependencies?  
  - [x] wakatime-cli
  - Please specify if there are any other dependencies below and in the README.md and package.xml files

### README.md
- [x] Is [Markdown][Markdown-link] friendly
- [x] Has a **Prerequisite** or **Dependencies** section
- [x] Links to a ``LICENCE`` file (Note: FreeCAD uses LGPL2.1, most people choose that or GPL3)
- [x] Has a Screenshot and/or a section for screenshots
- [ ] Links to a [FreeCAD Forum discussion thread][FreeCAD-Thread]

### Misc.
- [x] Do you keep a [changelog][Changelog]?  
  If so, where?  
- [x] Your Addon has an SVG based Logo? Using wakatime's png logo 
- [x] Your Addon has a [package.xml file][Package_Metadata]?
- [ ] Edit the FreeCAD [External Workbenches][FC-ExternalWB-Wiki] wiki page

### Housekeeping
- [x] Fixed typos in the code using [`codespell`][Codespell]. We recommend running: `codespell -q 3`
- [ ] We recommend linting python code with PEP8. See [coding conventions][Code-conventions] 

### Github Related
- [x] Does your repository use [Release tags][Github-Tags] ?
- [ ] Repository has [topics][Github-Topics] which include `FreeCAD` `Workbench` `Addon`  

**Note:** if your addon/workbench code doesn't reside on Github, please state the address where it is mentioned

 
[Markdown-link]: https://guides.github.com/features/mastering-markdown/
[FreeCAD-Thread]: https://forum.freecadweb.org/viewforum.php?f=8
[Changelog]: https://keepachangelog.com/en/1.0.0/
[Package_Metadata]: https://wiki.freecad.org/Package_Metadata
[FC-ExternalWB-Wiki]: https://www.freecadweb.org/wiki/External_workbenches
[Coding-conventions]: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Fem/coding_conventions.md#python-and-c
[Codespell]: https://github.com/codespell-project/codespell
[Github-Topics]: https://help.github.com/en/articles/classifying-your-repository-with-topics
[Github-Tags]: https://help.github.com/en/articles/creating-releases

luzpaz, 2025-01-31:
https://github.com/Pegoku/freecad-wakatime/issues/2

luzpaz, 2025-02-09:
The freecad name is actually 'FreeCAD'.  Can you rename the workbench appropriately ?

luzpaz, 2025-02-09:
Actually, I'd recommend the workbench be named 'FreeCAD-WakaTime'

PR #356 (2025-02-12): Added WakaTime workbench
Closes #349

Ref: https://github.com/Pegoku/FreeCAD-WakaTime/issues/2 & #355