FreeCAD/FreeCAD-addons #361: [ 🚮 ] Remove deprecated Python package ( dotenv )

PhoneDroid, 2025-03-13:
The `dotenv` has been deprecated.

[image]

[image]

### DotEnv
- [Registry](https://pypi.org/project/dotenv/)
- [GitHub](https://github.com/pedroburon/dotenv?tab=readme-ov-file)

---

Author recommends the `load_dotenv` library.

### LoadDotEnv

- [Registry](https://pypi.org/project/load-dotenv/)
- [GitHub](https://github.com/theskumar/python-dotenv)

---

### Affected Addons

I clones all the addon repositories in `FreeCAD/FreeCAD-addons` and  
searched for `dotenv` but it doesn't seem like any of them use it.

chennes, 2025-03-13:
@AmulyaParitosh should see this, I think it's the https://github.com/FreeCAD/FreeCAD-Component-Library-Addon that asked for it originally.

PhoneDroid, 2025-08-20:
They don't seem to be very active on GitHub  
recently, I'll try to contact them via Discord.

[image]