FreeCAD/FreeCAD-addons #390: assembly4 link in the addon manager is not working

muezabdalla, 2025-06-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

the link of the assembly4 addon give 404 error (even in the browser
I think we should switch to a fork of it
like this one:
https://github.com/oursland/FreeCAD_Assembly4?tab=readme-ov-file

### Full version info

```shell
OS: Void Linux (labwc:wlroots/wayland)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git) AppImage
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * A2plus 0.4.68
  * FreeCAD_Assembly4-master.zip
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * AnimationFreeCAD 1.0.0
  * ExplodedAssembly
  * CfdOF 1.30.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-06-23:
The original repository had `master` as the main repo.
In this new one, it should be `main`.

[image]

Do you guys know how do I fix this?

leoheck, 2025-06-23:
@muezabdalla, please, use this one if you want to install it manually for now.
https://github.com/leoheck/FreeCAD_Assembly4 (the main branch points to the 0.50.18)

leoheck, 2025-06-23:
Fixed my package.xml, let's see if this solves the problem.

leoheck, 2025-06-23:
Actually, the fix is here, https://github.com/FreeCAD/FreeCAD-addons/pull/391/files

hyarion, 2025-06-23:
@leoheck merged

leoheck, 2025-06-23:
Yeah, I think so. It is already updated on the Package Manager.

closed 2025-06-23 by hyarion

chennes, 2025-06-24:
Noting for the record that I don't expect the current Addon Manager to be able to *update* an existing ASM4 git clone to use the new repo. Zip-installed copies will probably work, but if you have git installed and want to be able to update ASM4 you will have to uninstall and reinstall. I'm working on a better solution, but it's not done yet.