FreeCAD/FreeCAD-addons #442: Help with the Assembly4.1

leoheck, 2025-09-21:
@chennes, quick question.

I am updating the forked Assembly4, naming it Assembly4.1
My package manager still sees it as `Assembly4`, and it shows the outdated version.

What should I do to be able to see the changes? 
This is the current status of the repo: https://github.com/leoheck/FreeCAD_Assembly4.1
Current settings on FreeCAD-addons seem correct.

This is how my AddonManager sees it:

[image]

This is the only big issue is that I have a version that is not quite standard. I am still thinking about what I will do with it.

[image]

[Off-topic bug] When I have the search bar filled, the selected item is not highlighted for some reason. You can see this in the image above, where the `Assembly4 (v0.50.18)` is selected.

leoheck, 2025-09-21:
Ah, never mind, it was just a matter of time.

[image]

How do I enable or configure that feature you were telling me to allow visibility of different branches?

PhoneDroid, 2025-09-22:
@leoheck Search for the Rocket workbench in the AddonCatalog for an example.

chennes, 2025-09-22:
> Ah, never mind, it was just a matter of time.

Just for the record, the process that generates that cache runs every six hours, so the delay you saw (if you are using the latest addon manager) was due to that lag. If you're using an older addon manager, the cache is local, and only regenerated on a schedule you set in the preferences.

leoheck, 2025-09-22:
> Rocket workbench

Nice, thanks. I was indeed looking for a workbench that I could use as a model.

[Off-topic] By the way, @chennes, the search is quite restrictive... 
It works, but it could also show results for matching individual words.
In this case, I pasted the `Rocket workbench term` and I got nothing. It could be better to show everything that matches with both words. The ones that match the whole thing should appear first, of course.
[image]

PhoneDroid, 2025-09-22:
Well it should just use fuzzy search.

chennes, 2025-09-22:
"just", he says! 😆

Right now it's a regex, so you can achieve what you want @leoheck but you need to do the regex dance. Implemented that way because it was easy, of course. I probably won't have time to get to it anytime soon, but feel free to add an Issue to the Addon Manager's [tracker](https://github.com/FreeCAD/AddonManager/issues) about search, and maybe someone will decide to pick it up.