FreeCAD/FreeCAD-addons #443: Save and restore Addon is missing from add-on manager

APEbbers, 2025-09-26:
Hi, 

With PR #376, the Save and restore Add-on was added to the available add-ons.
However now the add-on seems missing. Was there anything wrong with the add-on or is it removed by accident?

If possible I'd like to add this add-on again, because I use it with the RibbonUI.

Kind regards

Paul Ebbers

PhoneDroid, 2025-09-26:
Well it doesn't seem to be listed in the [Addon Catalog].

[Addon Catalog]: https://github.com/FreeCAD/FreeCAD-addons/blob/master/AddonCatalog.json

chennes, 2025-09-26:
Probably what happened was just bad timing: I think that's right when I was auto-generating the catalog from the modules file, there was probably a lag between when I did that work and when I pushed it, and this addon was added right then (and I didn't think about it when I merged the modules change). So  PR to fix the catalog would be appreciated! Sorry for the mixup.

APEbbers, 2025-09-26:
No worries, I've create PR #444  to fix the catalog